jsk-ros-pkg/jsk_roseus #208: conflicting targets in roseus.cmake

garaemon, 2015-01-26:
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/cmake/roseus.cmake#L64

```
CMake Error at /opt/ros/hydro/share/roseus/cmake/roseus.cmake:64 (add_custom_target):
  add_custom_target cannot create target "actionlib_generate_messages_roseus"
  because another target with the same name already exists.  The existing
  target is a custom target created in source directory
  "/home/travis/ros/ws_jsk_demos/src/jsk-ros-pkg/jsk_planning/task_compiler".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  /opt/ros/hydro/share/roseus/cmake/roseus.cmake:83 (generate_all_roseus_messages)
  /opt/ros/hydro/share/roseus/cmake/roseusConfig.cmake:190 (include)
  /opt/ros/hydro/share/catkin/cmake/catkinConfig.cmake:75 (find_package)
  jsk-ros-pkg/jsk_smart_apps/jsk_smart_gui/catkin.cmake:3 (find_package)
  jsk-ros-pkg/jsk_smart_apps/jsk_smart_gui/CMakeLists.txt:2 (include)
```

k-okada, 2015-01-26:
> Assigned #208 <https://github.com/jsk-ros-pkg/jsk_roseus/issues/208> to
> @k-okada <https://github.com/k-okada>.

sorry this is fixed in https://github.com/jsk-ros-pkg/jsk_roseus/pull/207,
you'll get this deb binary withtin 6 hours

garaemon, 2015-01-30:
solved

closed 2015-01-30 by garaemon