jsk-ros-pkg/jsk_roseus #209: cannot execute roseus

garaemon, 2015-01-26:
cannot run roseus
```
$ rosrun roseus roseus
[rospack] Error: stack/package euslisp not found

```

1. I installed euslisp/jskeus/roseus by setup.sh
2. I executed catkin build roseus
3. successfully compiled
4. source devel/setup.bash
5. cannot find euslisp packae
6. environment variables are:
```
$ echo $PATH
/home/leus/ros/hydro/devel/share/euslisp/jskeus/eus//Linux64/bin:/home/leus/ros/hydro/devel/bin:/opt/ros/hydro/bin:/home/leus/ros/hydro/devel/share/euslisp/jskeus/eus//Linux64/bin:/home/leus/ros/hydro/devel/lib/rtshell:/home/leus/ros/hydro/src/jsk-ros-pkg/jsk_roseus/euslisp/jskeus/eus/Linux64/bin:/home/leus/ros/hydro/src/jsk-ros-pkg/jsk_roseus/euslisp/jskeus/eus/Linux64/lib:/usr/lib/lightdm/lightdm:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/home/leus/ros/hydro/src/jsk-ros-pkg/jsk_roseus/roseus/bin:/home/leus/android-sdk-linux/tools:/home/leus/android-sdk-linux/platform-tools
$ echo $ROS_PACKAGE_PATH
/home/leus/ros/hydro/src/jsk-ros-pkg/jsk_roseus/geneus:/home/leus/ros/hydro/src/jsk-ros-pkg/jsk_roseus/roseus:/opt/ros/hydro/share:/opt/ros/hydro/stacks
```
7. simply executing `irteusgl` works

k-okada, 2015-01-26:
>    1. simply executing irteusgl works

please copy and paste the output of `catkin clean --all; catkin build`

garaemon, 2015-01-26:
https://gist.github.com/garaemon/ed980a700edc935adb4b

garaemon, 2015-01-26:
Can you get `rospack find euslisp`?

k-okada, 2015-01-26:
no, but rosrun euslisp eusgl works. humm, should we enable rospack find
euslisp?

garaemon, 2015-01-26:
https://github.com/tork-a/euslisp-release/blob/release/indigo/euslisp/package.xml#L40
this line

garaemon, 2015-01-26:
Why can you run rosrun euslisp without rospack find ??

garaemon, 2015-01-26:
catkin_find?
```
$ catkin_find euslisp
/home/leus/ros/hydro/devel/share/euslisp
```

I re-source bashrc and I can run `rosrun euslisp eusg`
```
$ rosrun euslisp eusg[rospack] Error: stack/package euslisp not found
configuring by "/home/leus/ros/hydro/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; l/par.l ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; fstringdouble 
EusLisp 9.10(891e20c) for Linux64 created on yoshino(Mon Jan 26 18:08:45 JST 2015)
1.eusg$ 
```

garaemon, 2015-01-26:
However, rosrun still not work...
```
$ rosrun roseus roseus
[rospack] Error: stack/package euslisp not found
```

k-okada, 2015-01-26:
howabout apt-get purge ros-hydro-euslisp ros-hydro-roseus?

garaemon, 2015-01-26:
roseus script uses rospack to 
1. find euslisp directory
2. find roseus directory

2 is OK. 1 is not good.

garaemon, 2015-01-26:
I do not install euslisp and roseus on my environment via apt

garaemon, 2015-01-26:
I think we need to replace `rospack` with `catkin_find`

```diff
diff --git a/roseus/bin/roseus b/roseus/bin/roseus
index 182f738..4c32800 100755
--- a/roseus/bin/roseus
+++ b/roseus/bin/roseus
@@ -1,7 +1,7 @@
 #!/bin/bash
 set -e
 
-EUSLISP_DIR=`rospack find euslisp`
+EUSLISP_DIR=`catkin_find --libexec --share euslisp`
 EUSLISP_EXE=`find -L $EUSLISP_DIR -type f -name irteusgl`
 ROSEUS_DIR=`rospack find roseus`
 ARG_STR=("(pushnew \"${ROSEUS_DIR}/euslisp/\" *load-path* :test #'equal)" \
```

garaemon, 2015-01-26:
https://github.com/jsk-ros-pkg/jsk_roseus/pull/210
Please check it

k-okada, 2015-01-26:
I'm super sorry, that my fault,
https://github.com/tork-a/euslisp-release/blob/release/indigo/euslisp/env-hooks/99.euslisp.sh.in
allow as to use EUSDIR, see #211

garaemon, 2015-01-26:
solved!

closed 2015-01-26 by garaemon