jsk-ros-pkg/jsk_roseus #221: generate-all-msg-srv.sh is deprecated?

furushchev, 2015-02-09:
When I launch some launch file that contains roseus scripts, error occures as below:

```bash
[c2-0]: process[lhand_snap-48]: started with pid [18470]
[c2-0]: process[rhand_snap-49]: started with pid [18500]
remote[c2-0]: b16") on euslisp((9.10 host08.storm.ros.org Sat Jan 24 05:53:25 UTC 2015  1.0.2))
[ERROR] [1423483377.570358104]: Could not find roseus messages under (/home/furushchev/.ros/roseus/hydro /home/furushchev/ros/hydro/devel/
share/roseus/ros /opt/ros/hydro/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh libccd
[ERROR] [1423483378.512428581]: Could not find roseus messages under (/home/furushchev/.ros/roseus/hydro /home/furushchev/ros/hydro/devel/
share/roseus/ros /opt/ros/hydro/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh opencv2
[ERROR] [1423483378.512654085]: Could not find roseus messages under (/home/furushchev/.ros/roseus/hydro /home/furushchev/ros/hydro/devel/
share/roseus/ros /opt/ros/hydro/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh op
```

So I tried to execute `rosrun roseus generate-all-msg-srv.sh`, which occurs another errors:

```bash
$ rosrun roseus generate-all-msg-srv.sh libccd
package:libccd -> /opt/ros/hydro/share/libccd
generating... 1 files with ALL=No, SHARED=No, COMPILE=No option
generating... 0/1
generating manifest... libccd
/home/furushchev/ros/hydro/src/jsk-ros-pkg/jsk_roseus/roseus/scripts/generate-all-msg-srv.sh: line 111: /opt/ros/hydro/share/geneus/scripts/genmanifest_eus: No such file or directory
ERROR in /opt/ros/hydro/share/libccd
[ERROR] occurred while processing /home/furushchev/ros/hydro/src/jsk-ros-pkg/jsk_roseus/roseus/scripts/generate-all-msg-srv.sh
/opt/ros/hydro/share/libccd
```

this may be because `jsk-ros-pkg/geneus` has not `genmsg_eus`, `gensrv_eus`, `genmanifest_eus` in scripts directory.
How do we generate missing messages?

garaemon, 2015-02-09:
I have same issue now

◉  ryohei

k-okada, 2015-02-09:
me too, just a moment.

k-okada, 2015-02-09:
#222 will fix this, if your gen_eus.py is not executable, fix manually until  https://github.com/jsk-ros-pkg/geneus/pull/3 is released

k-okada, 2015-02-09:
basically we do not have to use `generate-all-msg-srv.sh` scripts, if you write correct cmake,
the rule is:
- invoke your script from package that depends on roseus, depends means you call find_package(catkin REQUIRE COMPONENT .... roseus)
- all package you need in your euslisp code must be listed in find_package(catkin REQUIRE COMPONENT ...) or packages you need mest be depend on a package in find_package
- roseus in find_pakcage() must be the last element, so that roseus.cmake can find dependent packages.

if you have package that you can not generate necessary euslisp message, please let me know

garaemon, 2015-02-09:
Good explanation, let's add to README.md

✉︎ Ryohei Ueda

garaemon, 2015-02-10:
is it working?

I've updated geneus and roseus
```
$ rosrun roseus generate-all-msg-srv.sh rosgraph_msgs
package:rosgraph_msgs -> /opt/ros/hydro/share/rosgraph_msgs
generating... 1 files with ALL=No, COMPILE=No option
generating... 0/1
generating msg... /opt/ros/hydro/share/rosgraph_msgs/msg/Clock.msg
Traceback (most recent call last):
  File "/home/jskuser/ros/hydro_parent/src/geneus/src/geneus/geneus_main.py", line 82, in genmain
    retcode = generate_msg(options.package, args[1:], options.outdir, search_path)
  File "/home/jskuser/ros/hydro_parent/src/geneus/src/geneus/generate.py", line 715, in generate_msg
    generate_msg_from_spec(msg_context, spec, search_path, out_dir, pkg)
  File "/home/jskuser/ros/hydro_parent/src/geneus/src/geneus/generate.py", line 746, in generate_msg_from_spec
    msgs = msg_list(package, search_path, '.msg')
  File "/home/jskuser/ros/hydro_parent/src/geneus/src/geneus/generate.py", line 730, in msg_list
    dir_list = search_path[pkg]
KeyError: 'rosgraph_msgs'
/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_roseus/roseus/scripts/generate-all-msg-srv.sh: 行 76: ERROR:: コマンドが見つかりません
ERROR in /opt/ros/hydro/share/rosgraph_msgs
generating msg... /opt/ros/hydro/share/rosgraph_msgs/msg/Log.msg
Traceback (most recent call last):
  File "/home/jskuser/ros/hydro_parent/src/geneus/src/geneus/geneus_main.py", line 82, in genmain
    retcode = generate_msg(options.package, args[1:], options.outdir, search_path)
  File "/home/jskuser/ros/hydro_parent/src/geneus/src/geneus/generate.py", line 715, in generate_msg
    generate_msg_from_spec(msg_context, spec, search_path, out_dir, pkg)
  File "/home/jskuser/ros/hydro_parent/src/geneus/src/geneus/generate.py", line 743, in generate_msg_from_spec
    genmsg.msg_loader.load_depends(msg_context, spec, search_path)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 344, in load_depends
    return load_msg_depends(msg_context, spec, msg_search_path)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 313, in load_msg_depends
    depspec = load_msg_by_type(msg_context, resolved_type, search_path)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 119, in load_msg_by_type
    file_path = get_msg_file(package_name, base_type, search_path)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 78, in get_msg_file
    % (base_type, package, search_path))
MsgNotFound: Cannot locate message [Header]: unknown package [std_msgs] on search path [{}]
/home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_roseus/roseus/scripts/generate-all-msg-srv.sh: 行 76: ERROR:: コマンドが見つかりません
ERROR in /opt/ros/hydro/share/rosgraph_msgs
generating manifest... rosgraph_msgs
[ERROR] occurred while processing /home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_roseus/roseus/scripts/generate-all-msg-srv.sh
/opt/ros/hydro/share/rosgraph_msgs
/opt/ros/hydro/share/rosgraph_msgs

```

orikuma, 2015-02-11:
I met same problems.
```bash
$ rosrun roseus generate-all-msg-srv.sh                                                                                                                                                                                                                                                         [(git)-<master>][2-0]
package:nav_msgs -> /opt/ros/indigo/share/nav_msgs
package:tf -> /opt/ros/indigo/share/tf
package:std_msgs -> /opt/ros/indigo/share/std_msgs
package:geometry_msgs -> /opt/ros/indigo/share/geometry_msgs
package:sensor_msgs -> /opt/ros/indigo/share/sensor_msgs
package:sound_play -> /opt/ros/indigo/share/sound_play
package:actionlib_msgs -> /opt/ros/indigo/share/actionlib_msgs
package:visualization_msgs -> /opt/ros/indigo/share/visualization_msgs
package:pr2eus -> /home/iori/ros/indigo/src/jsk-ros-pkg/jsk_pr2eus/pr2eus
package:trajectory_msgs -> /opt/ros/indigo/share/trajectory_msgs
package:hrpsys_ros_bridge -> /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge
package:rosgraph_msgs -> /opt/ros/indigo/share/rosgraph_msgs
package:pr2_msgs -> /opt/ros/indigo/share/pr2_msgs
package:dynamic_reconfigure -> /opt/ros/indigo/share/dynamic_reconfigure
package:actionlib -> /opt/ros/indigo/share/actionlib
package:roscpp -> /opt/ros/indigo/share/roscpp
package:tf2_msgs -> /opt/ros/indigo/share/tf2_msgs
package:move_base_msgs -> /opt/ros/indigo/share/move_base_msgs
package:roseus -> /home/iori/ros/indigo/src/jsk-ros-pkg/jsk_roseus/roseus
package:bayesian_belief_networks -> /home/iori/ros/indigo/src/jsk-ros-pkg/jsk_common/bayesian_belief_networks
package:pr2_controllers_msgs -> /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_common/pr2_controllers_msgs
package:std_srvs -> /opt/ros/indigo/share/std_srvs
package:control_msgs -> /opt/ros/indigo/share/control_msgs
generating... 23 files with ALL=No, COMPILE=No option
generating... 0/23
generating msg... /opt/ros/indigo/share/nav_msgs/msg/GetMapActionGoal.msg
Traceback (most recent call last):
  File "/home/iori/ros/indigo/src/jsk-ros-pkg/jsk_roseus/geneus/src/geneus/geneus_main.py", line 82, in genmain
    retcode = generate_msg(options.package, args[1:], options.outdir, search_path)
  File "/home/iori/ros/indigo/src/jsk-ros-pkg/jsk_roseus/geneus/src/geneus/generate.py", line 715, in generate_msg
    generate_msg_from_spec(msg_context, spec, search_path, out_dir, pkg)
  File "/home/iori/ros/indigo/src/jsk-ros-pkg/jsk_roseus/geneus/src/geneus/generate.py", line 743, in generate_msg_from_spec
    genmsg.msg_loader.load_depends(msg_context, spec, search_path)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 344, in load_depends
    return load_msg_depends(msg_context, spec, msg_search_path)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 313, in load_msg_depends
    depspec = load_msg_by_type(msg_context, resolved_type, search_path)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 119, in load_msg_by_type
    file_path = get_msg_file(package_name, base_type, search_path)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 85, in get_msg_file
    (base_type, package, str(search_path[package])))
MsgNotFound: Cannot locate message [GetMapGoal] in package [nav_msgs] with paths [['/opt/ros/indigo/share/nav_msgs']]
ERROR:  Cannot locate message [GetMapGoal] in package [nav_msgs] with paths [['/opt/ros/indigo/share/nav_msgs']]
ERROR in /opt/ros/indigo/share/nav_msgs
generating msg... /opt/ros/indigo/share/nav_msgs/msg/GetMapFeedback.msg
generating msg... /opt/ros/indigo/share/nav_msgs/msg/GetMapActionResult.msg
Traceback (most recent call last):
  File "/home/iori/ros/indigo/src/jsk-ros-pkg/jsk_roseus/geneus/src/geneus/geneus_main.py", line 82, in genmain
    retcode = generate_msg(options.package, args[1:], options.outdir, search_path)
  File "/home/iori/ros/indigo/src/jsk-ros-pkg/jsk_roseus/geneus/src/geneus/generate.py", line 715, in generate_msg
    generate_msg_from_spec(msg_context, spec, search_path, out_dir, pkg)
  File "/home/iori/ros/indigo/src/jsk-ros-pkg/jsk_roseus/geneus/src/geneus/generate.py", line 743, in generate_msg_from_spec
    genmsg.msg_loader.load_depends(msg_context, spec, search_path)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 344, in load_depends
    return load_msg_depends(msg_context, spec, msg_search_path)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 313, in load_msg_depends
    depspec = load_msg_by_type(msg_context, resolved_type, search_path)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 119, in load_msg_by_type
    file_path = get_msg_file(package_name, base_type, search_path)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 85, in get_msg_file
    (base_type, package, str(search_path[package])))
MsgNotFound: Cannot locate message [GetMapResult] in package [nav_msgs] with paths [['/opt/ros/indigo/share/nav_msgs']]
ERROR:  Cannot locate message [GetMapResult] in package [nav_msgs] with paths [['/opt/ros/indigo/share/nav_msgs']]
ERROR in /opt/ros/indigo/share/nav_msgs
```

k-okada, 2015-02-11:
Please check this or https://github.com/jsk-ros-pkg/jsk_roseus/pull/224
--
◉ Kei Okada

k-okada, 2015-02-11:
Please check this PR https://github.com/jsk-ros-pkg/jsk_roseus/pull/224

--
◉ Kei Okada

2015/02/12 2:06、Kei Okada <kei.okada@gmail.com> のメッセージ:

> Please check this or https://github.com/jsk-ros-pkg/jsk_roseus/pull/224

k-okada, 2015-02-12:
this has been fixed #224

closed 2015-02-12 by k-okada