jsk-ros-pkg/jsk_roseus #234: rosrun roseus roseus fails.

aginika, 2015-02-14:
Does this happen only in my environment? This is using only apt-installed package.

`rosrun roseus roseus` style executing is deprecated?
`roseus` is the right way?
If so, we need to replace all the launch file to modify the part of  `type="roseus" pkg="roseus" ` ?

```
[rosrun] Couldn't find executable named roseus below /opt/ros/hydro/share/roseus
[rosrun] Found the following, but they're either not files,
[rosrun] or not executable:
[rosrun]   /opt/ros/hydro/share/roseus
[rosrun]   /opt/ros/hydro/share/roseus/ros/roseus

```

garaemon, 2015-02-16:
roseus in installed to `${CATKIN_GLOBAL_BIN_DESTINATION}` and rosrun should work.
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/CMakeLists.txt#L129

Please check `/opt/ros/hydro/{lib,share}/roseus` directory

aginika, 2015-02-17:
There was roseus binary.
But It can't execute...

```
$/opt/ros/hydro/lib/roseus$ ls
roseus
$:/opt/ros/hydro/share/roseus$ ls
cmake  euslisp  msg  package.xml  ros  rospack_nosubdirs  scripts  srv  test
```

garaemon, 2015-02-17:
check permission of the script

aginika, 2015-02-17:
```
install(PROGRAMS bin/roseus
  DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION})
```
This will install roseus binary to /opt/ros/indigo/bin.

I think `rosrun roseus` complition seems to try to find roseus binary in roseus package directory, but the bin directory is move to /opt/ros/indigo/bin , so It seems to fail.

aginika, 2015-02-17:
Sorry, you direct the below line.
```
install(PROGRAMS bin/roseus
  DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION})
```
but I coulldn't find the bin directory in my /opt/ros/indigo/share/roseus....

k-okada, 2015-02-17:
rosues is installed under /opt/ros/indigo/lib/roseus and other package like
tf is installed their program under lib/tf too, so something wrong, please
find the reason

aginika, 2015-02-17:
@k-kimura Do you also fail? Only in my environment?

k-okada, 2015-02-17:
it was failed,but after sourcing /opt/ros/indigo/setup.bash, now it fixed,
it seems rosrun checks pkg-config or something to find libexec path, check
using `-d` option of rosrun.

aginika, 2015-02-17:
rosrun will find the directroy as below.
1. try to find the libexec or share from the top overlay workspace and if there aren't directory, that will  keep searching until the redirectory found. But if found in first workspace, the search will end.
2. if not found in all workspace, next try to find the `rospack` directory.

This time the problem was the 1st way.

In 1st way,
This below command will find the target directory until the directory is found.
```
$ catkin_find --without-underlays --libexec --share roseus 2> /dev/null
/home/baxter/ros/indigo/devel/share/roseus
```
If the `--without-underlays` is true, the only first directory will be used.
So, if we remove this option, the command result will be
```
/home/baxter/ros/indigo/devel/share/roseus
/opt/ros/indigo/lib/roseus
/opt/ros/indigo/share/roseus
```

When we do `catkin build` ,for example, only for jsk_demos , the roseus directory will be made in devel/share/roseus even if roseus is installed in /opt/ros/indigo/share/roseus.

So rosrun can't find the binary in  1st way.
And 2nd way only try to find rospack directory.

So both 1st and 2nd way can't find the /opt/ros/indigo/lib/roseus.

How should we do?
- don't create directory in devel/share/roseus
- change `rosrun` to remove --witout-overlay option from catkin_find
- add another 3rd search method to rosrun to search in all workspace( add )

closed 2015-02-17 by aginika