jsk-ros-pkg/jsk_roseus #237: [feature request] Support timer

garaemon, 2015-02-17:
like ros::Timer in roscpp
http://wiki.ros.org/roscpp/Overview/Timers

garaemon, 2015-02-17:
Of course, we can do it like
```lisp
(while (ros::ok)
  (ros::sleep)
  (ros::spin-once)
  (proc))
```

k-okada, 2015-02-17:
> Of course, we can do it like
>
> (while (ros::ok)
>   (ros::sleep)
>   (ros::spin-once)
>   (proc))

humm, I didn't know this feature, rospy also have same function
http://wiki.ros.org/rospy/Overview/Time#Timer, ok I'll try.

garaemon, 2015-02-17:
It's a really good feature.
For example, I use it to send UDP packets periodically
https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk_network_tools/scripts/silverhammer_highspeed_streamer.py#L39

and publish diagnostic information periodically
https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk_topic_tools/src/timered_diagnostic_updater.cpp#L45

(About diagnostic information, I know there is better way to do that with official diagnostic_updater)

We can write "background job" with timer and If we can implement it without threads (for example add simple hook before ros::spin-once in roseus.cpp), roseus can be almost concurrent.