jsk-ros-pkg/jsk_roseus #251: segmentation fault: in callback function when using actionlib

wkentaro, 2015-02-21:
When using actionlib and roseus, I had segmentation fault in its callback function, and resolved it by passing arguments to callback function.

I mean, in the sample code:
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/test/fibonacci-server.l

from
```lisp
(setq s (instance ros::simple-action-server :init
                  "/fibonacci" actionlib_tutorials::FibonacciAction
                  :execute-cb #'fibonacci-execute-cb))
```
to
```lisp
(setq  s (instance ros::simple-action-server :init
                  "/fibonacci" actionlib_tutorials::FibonacciAction
                  :execute-cb #'(lambda (server goal) (fibonacci-execute-cb server goal)))
```

This segmentation fault doesn't always happen, and **the sample code has no problem**. In my project, however, I had it. 

Actually I don't know the fundamental cause but in my case it is resolved with upper change. As reference my change log is:
https://github.com/wkentaro/2014-semi/commit/ac8dddf9cafd37212917abbddb1f42abe25894b7#diff-2a23e2389483ccb4c96d7f07ba40062eR58

garaemon, 2015-02-21:
To put it simply, you cannot use `#'(lambda (...) ...)` form in `instance` because euslisp does not support closure (Technically, you can use [deligation pattern](http://en.wikipedia.org/wiki/Delegation_pattern) for this problem).

In order to understand this issue perfectly, you need to know closure, free variables, let frame, macro expansion. It's a good lesson to understand how software works.

wkentaro, 2015-02-22:
Thanks to your introduction, I read some documents of these. I think I understood your point.
Does the code `#('lambda (...) ...)` problem, if I don't intend to use free variables outside `(lambda (...) ...)` scope?

In the actionlib.l, 
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/euslisp/actionlib.l#L233
it uses `(funcall exeucte-cb self goal)` so `:execute-cb #'execute-cb` in `instance` should be okay. However, in my case it raises segmentation fault.
×`:execute-cb #'execute-cb`
◯ ```:execute-cb `execute-cb```
◯ ```:execute-cb #'(lambda (server goal) (execute-cb server goal)```
◯ ```:execute-cb `(lambda (server goal) (execute-cb server goal))```

garaemon, 2015-02-22:
> Does the code #('lambda (...) ...) problem, if I don't intend to use free variables outside (lambda (...) ...) scope?

It' a deep world of euslisp.

* euslisp SHOULD be able to call lambda form without free variables.
* euslisp CANNOT call lambda form which can have free variables outside of its parent environment.

It means that euslisp cannot call lambda function which is created in another let form.

On the other hand, euslip can `funcall` symbol and in that case euslip evaluate the symbol as function when calling `funcall`.
https://github.com/euslisp/EusLisp/blob/master/lisp/c/eval.c#L1072
```lisp
(funcall #'+ 1 2)
; => 3

(funcall '+ 1 2) 
; => 3
```

And, euslisp can call closure even if the environment pointed by closure is on stack.
```lisp
(let ((var 1))
  (let ((f #'(lambda (x) var)))    ; closure include free variable!
    (print (funcall f 100)))) ; call inside of environment 
; => 1
(funcall (let ((var 1)) ; call out side of environment
           (let ((f #'(lambda (x) var)))    ; closure!
             f)) 1)
; => SEGV
```

garaemon, 2015-02-22:
```lisp
(funcall (let ((var 1))
           (let ((lambda '(lambda (x) var)))    ; closure!
             lambda)) 1)
; => undefined variable var!
```

Because the code is evaluated as 
```lisp
(funcall (eval '(lambda (x) var)) 1)
```

And when euslisp evaluate the lambda, there is no such variable like `var`

garaemon, 2015-02-22:
And let us assume the code like followings:

```lisp
(let ((a 1)) ;; let-a
  (let ((b 2)) ;; let-b
     (let ((c 3)) ;; let-c
   )))
```

Stack of environment in euslisp evaluator is:

|nest depth|nickname|parent environment|
|---|---|---|
|0|global scope| none|
|1|let-a|global|
|2|let-b|let-a|
|3|let-c|let-b|

Euslisp can call lambda which created in parent environment.

`defun` always create lambda in global scope so that you can call `defun-ed` functions in any let scopes.

However if you have lambda craeted in `let-b`, you can call it in `let-b` and `let-c` scope but you cannot call it in `let-a` and `global scope`

garaemon, 2015-02-22:
[This article](http://practical-scheme.net/gauche/memo-stack-j.html) may help you to understand stack VM.

wkentaro, 2015-02-27:
Thanks. I got to know some of the characteristics of euslisp and what I should keep in mind when using it.

k-okada, 2015-03-13:
sorry for super late response, but did you said that 
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/test/fibonacci-server.l 
does not working? and we need to rewrite as ` :execute-cb #'(lambda (server 
goal) (fibonacci-execute-cb server goal)` ??

wkentaro, 2015-03-14:
No, `fibonacci-server.l` has no problem. 
But in my code I had to write as `:execute-cb #'(lambda (server
goal) (fibonacci-execute-cb server goal)`.

k-okada, 2015-03-14:
may be you can write as :execute-cb #'send self :execute-cb,
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/test/listener.l#L28

wkentaro, 2015-03-19:
In this code, `:execute-cb #'execute-cb` raises unexpected error.
(this is why I opened this issue)
https://github.com/wkentaro/2014-semi/commit/ac8dddf9cafd37212917abbddb1f42abe25894b7#diff-2a23e2389483ccb4c96d7f07ba40062eR58

and
In this code 
`:execute-cb #'send self :execute-cb` also raise other error message.
It seems that in instance block of `ros::simple-action-server` we can't use `:execute-cb #'send self :execute-cb`.
https://github.com/wkentaro/2014-semi/commit/c583794fd22a974cfb5261555d94d505d25b2284#diff-92fac444fd3df4007e9776ba5f97ac76R29

garaemon, 2015-06-12:
`self` is problem, self cannot be resolved in global scope. the callback is just evaled by `eval` special form on the top level scope

k-okada, 2015-06-15:
I see, please describe that on
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/roseus.cpp#L1555
so that we can see that note on
http://euslisp-docs.readthedocs.org/en/latest/roseus/roseus/#rossubscribe >
@wkentaro

wkentaro, 2015-06-15:
I have never encountered this problem when using subscriber but actionlib.
Is that really impossible to use `self` in callback function of subscriber, and [this](https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/test/listener.l#L28) is problem?
In that case, I have no idea how to declare subscription using an instance method as callback function.. :(

garaemon, 2015-06-16:
`:execute-cb #'send self ...`
  This syntax is simply not supported. https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/euslisp/actionlib.l#L338

garaemon, 2015-06-16:
In subscription callback function, `#'send self :foo` is evaluated before `ros::subscribe` is called

wkentaro, 2015-06-16:
>
I see, please describe that on
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/roseus.cpp#L1555
so that we can see that note on
http://euslisp-docs.readthedocs.org/en/latest/roseus/roseus/#rossubscribe >
@wkentaro

I see, so where should I describe about that? @k-okada 
It seems that https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/roseus.cpp#L1555 has nothing to change.

furushchev, 2016-10-23:
@wkentaro Is this issue solved?