jsk-ros-pkg/jsk_roseus #257: wrong order in manifest.l

furushchev, 2015-03-03:
some package are built manifest.l with wrong order.

for example, `catkin build jsk_interactive_marker`

```lisp
# manifest.l
;;
;; DO NOT EDIT THIS FILE
;;
(ros::load-ros-package "audio_common_msgs")
(ros::load-ros-package "stereo_msgs")
(ros::load-ros-package "view_controller_msgs")
(ros::load-ros-package "actionlib_msgs")
(ros::load-ros-package "trajectory_msgs")
(ros::load-ros-package "dynamic_tf_publisher")
(ros::load-ros-package "tf2_msgs")
(ros::load-ros-package "nav_msgs")
(ros::load-ros-package "roscpp")
(ros::load-ros-package "jsk_interactive_marker")   ;; duplicate decrelation? if comment out here, it works.
(ros::load-ros-package "actionlib")
(ros::load-ros-package "diagnostic_msgs")
(ros::load-ros-package "topic_tools")
(ros::load-ros-package "jsk_footstep_msgs")
(ros::load-ros-package "control_msgs")
(ros::load-ros-package "pr2_msgs")
(ros::load-ros-package "ml_classifiers")
(ros::load-ros-package "visualization_msgs")
(ros::load-ros-package "tf")
(ros::load-ros-package "jsk_recognition_msgs")
(ros::load-ros-package "jsk_topic_tools")
(ros::load-ros-package "nodelet")
(ros::load-ros-package "pcl_msgs")
(ros::load-ros-package "people_msgs")
(ros::load-ros-package "rosgraph_msgs")
(ros::load-ros-package "shape_msgs")
(ros::load-ros-package "std_msgs")
(ros::load-ros-package "moveit_msgs")
(ros::load-ros-package "sensor_msgs")
(ros::load-ros-package "octomap_msgs")
(ros::load-ros-package "object_recognition_msgs")
(ros::load-ros-package "map_msgs")
(ros::load-ros-package "move_base_msgs")
(ros::load-ros-package "laser_assembler")
(ros::load-ros-package "jsk_hark_msgs")
(ros::load-ros-package "image_view2")
(ros::load-ros-package "pr2_controllers_msgs")
(ros::load-ros-package "dynamic_reconfigure")
(ros::load-ros-package "roseus")
(ros::load-ros-package "std_srvs")
(ros::load-ros-package "sound_play")
(ros::load-ros-package "jsk_rviz_plugins")
(ros::load-ros-package "geometry_msgs")
(ros::load-ros-package "jsk_gui_msgs")
(ros::load-ros-package "jsk_pcl_ros")
(ros::load-ros-package "roscpp_tutorials")
(ros::load-ros-package "bond")
(ros::load-ros-package "jsk_interactive_marker")
```

And I also wonder packages written in `generate_messages` at CMakeLists.txt is not the same as ones at the top of `manifest.l`:

```bash
# CMakeLists.txt
generate_messages(
  DEPENDENCIES geometry_msgs jsk_footstep_msgs visualization_msgs jsk_recognition_msgs
)
```

any idea?

k-okada, 2015-03-03:
Please create test code that fails in this situation in roseus

k-okada, 2015-03-03:
If I remember correctly, load-ros-package should work in any situation. It
does not assume some package is already loaded.

2015年3月4日水曜日、Kei Okada<k-okada@jsk.t.u-tokyo.ac.jp>さんは書きました:

> Please create test code that fails in this situation in roseus
>
> 2015年3月3日火曜日、Furushchev<notifications@github.com
>
>> some package are built manifest.l with wrong order.
>>
>> for example, catkin build jsk_interactive_marker
>>
>> # manifest.l;;;; DO NOT EDIT THIS FILE;;
>> (ros::load-ros-package "audio_common_msgs")
>> (ros::load-ros-package "stereo_msgs")
>> (ros::load-ros-package "view_controller_msgs")
>> (ros::load-ros-package "actionlib_msgs")
>> (ros::load-ros-package "trajectory_msgs")
>> (ros::load-ros-package "dynamic_tf_publisher")
>> (ros::load-ros-package "tf2_msgs")
>> (ros::load-ros-package "nav_msgs")
>> (ros::load-ros-package "roscpp")
>> (ros::load-ros-package "jsk_interactive_marker")   ;; duplicate decrelation? if comment out here, it works.
>> (ros::load-ros-package "actionlib")
>> (ros::load-ros-package "diagnostic_msgs")
>> (ros::load-ros-package "topic_tools")
>> (ros::load-ros-package "jsk_footstep_msgs")
>> (ros::load-ros-package "control_msgs")
>> (ros::load-ros-package "pr2_msgs")
>> (ros::load-ros-package "ml_classifiers")
>> (ros::load-ros-package "visualization_msgs")
>> (ros::load-ros-package "tf")
>> (ros::load-ros-package "jsk_recognition_msgs")
>> (ros::load-ros-package "jsk_topic_tools")
>> (ros::load-ros-package "nodelet")
>> (ros::load-ros-package "pcl_msgs")
>> (ros::load-ros-package "people_msgs")
>> (ros::load-ros-package "rosgraph_msgs")
>> (ros::load-ros-package "shape_msgs")
>> (ros::load-ros-package "std_msgs")
>> (ros::load-ros-package "moveit_msgs")
>> (ros::load-ros-package "sensor_msgs")
>> (ros::load-ros-package "octomap_msgs")
>> (ros::load-ros-package "object_recognition_msgs")
>> (ros::load-ros-package "map_msgs")
>> (ros::load-ros-package "move_base_msgs")
>> (ros::load-ros-package "laser_assembler")
>> (ros::load-ros-package "jsk_hark_msgs")
>> (ros::load-ros-package "image_view2")
>> (ros::load-ros-package "pr2_controllers_msgs")
>> (ros::load-ros-package "dynamic_reconfigure")
>> (ros::load-ros-package "roseus")
>> (ros::load-ros-package "std_srvs")
>> (ros::load-ros-package "sound_play")
>> (ros::load-ros-package "jsk_rviz_plugins")
>> (ros::load-ros-package "geometry_msgs")
>> (ros::load-ros-package "jsk_gui_msgs")
>> (ros::load-ros-package "jsk_pcl_ros")
>> (ros::load-ros-package "roscpp_tutorials")
>> (ros::load-ros-package "bond")
>> (ros::load-ros-package "jsk_interactive_marker")
>>
>> And I also wonder packages written in generate_messages at
>> CMakeLists.txt is not the same as ones at the top of manifest.l:
>>
>> # CMakeLists.txt
>> generate_messages(
>>   DEPENDENCIES geometry_msgs jsk_footstep_msgs visualization_msgs jsk_recognition_msgs
>> )
>>
>> any idea?
>>
>>
>
>
> ◉ Kei Okada

wkentaro, 2015-05-10:
I think this is fixed by https://github.com/jsk-ros-pkg/geneus/pull/18 .

closed 2015-05-11 by k-okada