jsk-ros-pkg/jsk_roseus #260: How to handle irtviewers and simulators in multiple nodes which use roseus?

wkentaro, 2015-03-10:
In my project, I'm using roseus in multiple nodes to make robot's motion, and I launch these nodes with a launch file. The problem is that in each roseus programs run `(objects (list *robot*))` and too many windows will open.
I want the program to open simulator and irtviewer windows 

* only once and one window when they are launched

or

* each time when called from other programs and close the windows after that

k-okada, 2015-03-13:
each roseus node has different instance of `*robot*` so sharing window over multiple roseus may difficult, my suggestion is to unset `DISPLAY` environment variable to disable windows as follows:
```
<launch>
  <arg name="GUI" default="true" />
  <env name="DISPLAY" value="" unless="$(arg GUI)"/>
  <node type="roseus" pkg="roseus" name="roseus"
        args="/home/k-okada/hoga.l" />
</launch>
```
and call
```
roslaunch hoga.launch  GUI:=false
```
or you may better to use
```
  <arg name="DEBUG" default="true" />
  <env name="DISPLAY" value="" if="$(arg DEBUG)"/>
```

wkentaro, 2015-03-14:
Thanks a lot! It's what I'd like to do.

closed 2015-03-16 by wkentaro