jsk-ros-pkg/jsk_roseus #273: ros message constant does not match value in `case`

furushchev, 2015-04-18:
This is just a question.

comparing ros message constant using `case` does not work.

```lisp
(ros::roseus-add-msgs "actionlib_msgs")

(setq a (send (instance actionlib_msgs::GoalStatus :init :status actionlib_msgs::GoalStatus::*lost*) :status))

(case a
  (actionlib_msgs::GoalStatus::*lost*
   (warn "hogehogehogehoge~%"))
  (t (warn "fugafuga~%")))

(if (eq a actionlib_msgs::GoalStatus::*lost*)
    (warn "eq!~%")
  (warn "not eq~%"))
```

This should prints:

```
fugafuga
eq!
```

Is this correct behavior?

k-okada, 2015-04-21:
```
(defconstant daa 0)
(setq dbb (round daa))
(print (list 'eq (eq 0 daa)))
(print (list 'eq (eq daa dbb)))
(case daa
      (daa (pirnt "case ok"))
      (t   (print "case ng")))
(case dbb
      (daa (pirnt "case ok"))
      (t   (print "case ng")))
(print (macroexpand '(case daa
                           (daa (pirnt "case ok"))
                           (t   (print "case ng")))))
```
outputs
```
(eq t)
(eq t)
"case ng"
"case ng"
(let ((#:case373 daa)) (if (eq #:case373 'daa) (pirnt "case ok") (if t (print "case ng") nil nil) nil))

```

it this bug? need to check on clisp

k-okada, 2015-04-22:
sbcl and clisp outputs similar results.
```
* k-okada@kokada-t440s:/tmp$ sbcl 
This is SBCL 1.1.14.debian, an implementation of ANSI Common Lisp.
More information about SBCL is available at <http://www.sbcl.org/>.

SBCL is free software, provided as is, with absolutely no warranty.
It is mostly in the public domain; some portions are provided under
BSD-style licenses.  See the CREDITS and COPYING files in the
distribution for more information.
* (load "hoge.l")

; file: /tmp/hoge.l
; in: SETQ DBB
;     (SETQ DBB (ROUND DAA))
; 
; caught WARNING:
;   undefined variable: DBB
; 
; compilation unit finished
;   Undefined variable:
;     DBB
;   caught 1 WARNING condition

(EQ T) 

; in: CASE DAA
;     (PIRNT "case ok")
; 
; caught STYLE-WARNING:
;   undefined function: PIRNT
; 
; compilation unit finished
;   Undefined function:
;     PIRNT
;   caught 1 STYLE-WARNING condition
(EQ T) 

; in: CASE DBB
;     (CASE DBB (DAA (PIRNT "case ok")) (T (PRINT "case ng")))
; --> PROGN 
; ==>
;   (LET ((#:G608 DBB))
;     (DECLARE (IGNORABLE #:G608))
;     (COND ((EQL #:G608 'DAA) NIL (PIRNT "case ok")) (T NIL (PRINT "case ng"))))
; 
; caught WARNING:
;   undefined variable: DBB

;     (PIRNT "case ok")
; 
; caught STYLE-WARNING:
;   undefined function: PIRNT
; 
; compilation unit finished
;   Undefined function:
;     PIRNT
;   Undefined variable:
;     DBB
;   caught 1 WARNING condition
;   caught 1 STYLE-WARNING condition
"case ng" 
"case ng" 
(LET ((#:G609 DAA))
  (DECLARE (IGNORABLE #:G609))
  (COND ((EQL #:G609 'DAA) NIL (PIRNT "case ok")) (T NIL (PRINT "case ng")))) 
T
* k-okada@kokada-t440s:/tmp$ clisp hoge.l

(EQ T) 
(EQ T) 
"case ng" 
"case ng" 
(LET ((#:CASE-KEY-3211 DAA)) (COND ((EQL #:CASE-KEY-3211 'DAA) (PIRNT "case ok")) (T (PRINT "case ng")))) 
```

furushchev, 2015-09-27:
@k-okada Sorry for very late response.
OK, I'll try to use `cond` in case of comparing constants.

closed 2015-09-27 by furushchev