jsk-ros-pkg/jsk_roseus #275: Cannot use ros communication

snozawa, 2015-04-20:
I tried
```
rtmlaunch hrpsys_ros_bridge_tutorials samplerobot.launch
```
and 
```
irteusgl$ (progn (load "package://hrpsys_ros_bridge_tutorials/euslisp/samplerobot-interface.l") (samplerobot-init))
irteusgl$ (send *ri* :autobalancerservice_setGaitGeneratorParam)
```
on latest geneus.

Then, I found that i could not use roseus communication as the following errors:
```
irteusgl$ (send *ri* :autobalancerservice_setGaitGeneratorParam)
;; Segmentation Fault.
;; in (write-long (length ros::_toe_heel_phase_ratio) ros::s)
;; You are still in a signal handler.
;;Try reset or throw to upper level as soon as possible.
;; code=608698224 x=2447fe40 addr=0
Fatal: (btrace)
signal=11 to thread 0, 
(write-long (length ros::_toe_heel_phase_ratio) ros::s)
(let ((ros::s (if ros::strm ros::strm (make-string-output-stream (send self :serialization-length))))) (system:poke ros::_default_step_time (send ros::s :buffer) (send ros::s :count) :double) (incf (stream-count ros::s) 8) (system:poke ros::_default\
_step_height (send ros::s :buffer) (send ros::s :count) :double) (incf (stream-count ros::s) 8) (system:poke ros::_default_double_support_ratio (send ros::s :buffer) (send ros::s :count) :double) (incf (stream-count ros::s) 8) (write-long (length ro\
s::_stride_parameter) ros::s) (dotimes (ros::i 4) (system:poke (elt ros::_stride_parameter ros::i) (send ros::s :buffer) (send ros::s :count) :double) (incf (stream-count ros::s) 8)) (progn (system:poke ros::_default_orbit_type (send ros::s :buffer)\
 (send ros::s :count) :long) (incf (stream-count ros::s) 8)) (system:poke ros::_swing_trajectory_delay_time_offset (send ros::s :buffer) (send ros::s :count) :double) (incf (stream-count ros::s) 8) (write-long (length ros::_stair_trajectory_way_poin\
t_offset) ros::s) (dotimes (ros::i 3) (system:poke (elt ros::_stair_trajectory_way_point_offset ros::i) (send ros::s :buffer) (send ros::s :count) :double) (incf (stream-count ros::s) 8)) (system:poke ros::_gravitational_acceleration (send ros::s :b\
uffer) (send ros::s :count) :double) (incf (stream-count ros::s) 8) (system:poke ros::_toe_pos_offset_x (send ros::s :buffer) (send ros::s :count) :double) (incf (stream-count ros::s) 8) (system:poke ros::_heel_pos_offset_x (send ros::s :buffer) (se\
nd ros::s :count) :double) (incf (stream-count ros::s) 8) (system:poke ros::_toe_zmp_offset_x (send ros::s :buffer) (send ros::s :count) :double) (incf (stream-count ros::s) 8) (system:poke ros::_heel_zmp_offset_x (send ros::s :buffer) (send ros::s \
:count) :double) (incf (stream-count ros::s) 8) (system:poke ros::_toe_angle (send ros::s :buffer) (send ros::s :count) :double) (incf (stream-count ros::s) 8) (system:poke ros::_heel_angle (send ros::s :buffer) (send ros::s :count) :double) (incf (\
stream-count ros::s) 8) (write-long (length ros::_toe_heel_phase_ratio) ros::s) (dotimes (ros::i (length ros::_toe_heel_phase_ratio)) (system:poke (elt ros::_toe_heel_phase_ratio ros::i) (send ros::s :buffer) (send ros::s :count) :double) (incf (str\
eam-count ros::s) 8)) (if ros::_use_toe_joint (write-byte -1 ros::s) (write-byte 0 ros::s)) (if (null ros::strm) (get-output-stream-string ros::s)))
(send ros::_i_param :serialize ros::s)
(let ((ros::s (if ros::strm ros::strm (make-string-output-stream (send self :serialization-length))))) (send ros::_i_param :serialize ros::s) (if (null ros::strm) (get-output-stream-string ros::s)))
(ros::service-call "/AutoBalancerServiceROSBridge/setGaitGeneratorParam" (instance #<metaclass #X621a688 hrpsys_ros_bridge::openhrp_autobalancerservice_setgaitgeneratorparamrequest> :init :i_param ros::__i_param))
(send *ri* :autobalancerservice_setgaitgeneratorparam)
#<compiled-code #X54964f8>
#<compiled-code #X54964f8>
```

I reverted geneus at 24c7a101c54a67b4e744bcc77603e6911e1140fd and I could execute the above-mentioned commands without errors.

k-okada, 2015-04-20:
@snozawa, please paste the results of 
```
rossrv show hrpsys_ros_bridge/OpenHRP_AutoBalancerService_getGaitGeneratorParam
```

@furushchev , create test code and fix them asap

snozawa, 2015-04-20:
I executed command for both getter and setter:
```
$ rossrv show hrpsys_ros_bridge/OpenHRP_AutoBalancerService_getGaitGeneratorParam
--
bool operation_return
hrpsys_ros_bridge/OpenHRP_AutoBalancerService_GaitGeneratorParam i_param
  float64 default_step_time
  float64 default_step_height
  float64 default_double_support_ratio
  float64[4] stride_parameter
  int64 default_orbit_type
  float64 swing_trajectory_delay_time_offset
  float64[3] stair_trajectory_way_point_offset
  float64 gravitational_acceleration
  float64 toe_pos_offset_x
  float64 heel_pos_offset_x
  float64 toe_zmp_offset_x
  float64 heel_zmp_offset_x
  float64 toe_angle
  float64 heel_angle
  float64[] toe_heel_phase_ratio
  bool use_toe_joint

$ rossrv show hrpsys_ros_bridge/OpenHRP_AutoBalancerService_setGaitGeneratorParam
hrpsys_ros_bridge/OpenHRP_AutoBalancerService_GaitGeneratorParam i_param
  float64 default_step_time
  float64 default_step_height
  float64 default_double_support_ratio
  float64[4] stride_parameter
  int64 default_orbit_type
  float64 swing_trajectory_delay_time_offset
  float64[3] stair_trajectory_way_point_offset
  float64 gravitational_acceleration
  float64 toe_pos_offset_x
  float64 heel_pos_offset_x
  float64 toe_zmp_offset_x
  float64 heel_zmp_offset_x
  float64 toe_angle
  float64 heel_angle
  float64[] toe_heel_phase_ratio
  bool use_toe_joint
---
bool operation_return
```

k-okada, 2015-04-20:
sorry for that

I created complete test code at
https://github.com/jsk-ros-pkg/jsk_roseus/pull/276
and fixed version of geneus -> https://github.com/jsk-ros-pkg/geneus/pull/22

snozawa, 2015-04-20:
Thanks a lot!

I checked the PR.
> geneus -> https://github.com/jsk-ros-pkg/geneus/pull/22

However, loading of generated `.l` fails
```
eustf roseus_c_util 1.irteusgl$ load "jaxon-interface.l 
/home/leus/ros/hydro_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: unbound variable self in (apply #'load-org-for-ros fullname args)
2.E1-ROS:irteusgl$ btrace
euserror
euserror
(apply #'load-org-for-ros fullname args)
(apply #'load-org-for-ros fullname args)
(let ((fullname fname)) (if (substringp "package://" fname) (setq fullname (resolve-ros-path fname))) (if (null fullname) (error "file ~s not found" fname)) (apply #'load-org-for-ros fullname args))
(require modname dirfile)
(require modname dirfile)
(let ((dirfile (concatenate string dir file)) (modname (intern (string-upcase (format nil "~A/~A" pkg (pathname-name file))) *keyword-package*))) (require modname dirfile) (when *compile-message* (in-package "ROS") (let* ((fname (pathname-name dirfile)) (old-module (find (format ni\
l "~A_~A" pkg fname) *loaded-modules* :key #'lisp::load-module-file-name :test #'equal))) (unless old-module (let ((ros-home (ros-home-dir))) (compiler::compile-file-if-src-newer-so dirfile :out-prefix (format nil "~A_" pkg) :output-dir (format nil "~A/roseus/~A/" ros-home (unix:ge\
tenv "ROS_DISTRO")) :clear-files t) (load (format nil "~A_~A.so" pkg fname) :entry (format nil "___~A" fname))))) (in-package "USER")))
(progn (let ((dirfile (concatenate string dir file)) (modname (intern (string-upcase (format nil "~A/~A" pkg (pathname-name file))) *keyword-package*))) (require modname dirfile) (when *compile-message* (in-package "ROS") (let* ((fname (pathname-name dirfile)) (old-module (find (fo\
rmat nil "~A_~A" pkg fname) *loaded-modules* :key #'lisp::load-module-file-name :test #'equal))) (unless old-module (let ((ros-home (ros-home-dir))) (compiler::compile-file-if-src-newer-so dirfile :out-prefix (format nil "~A_" pkg) :output-dir (format nil "~A/roseus/~A/" ros-home (\
unix:getenv "ROS_DISTRO")) :clear-files t) (load (format nil "~A_~A.so" pkg fname) :entry (format nil "___~A" fname))))) (in-package "USER"))))
t
3.E1-ROS:irteusgl$ ,fullname
"/home/leus/ros/hydro/devel/share/roseus/ros/hrpsys_ros_bridge/msg/OpenHRP_OGMap3D.l"
4.E1-ROS:irteusgl$ ls /home/leus/ros/hydro/devel/share/roseus/ros/hrpsys_ros_bridge/msg/OpenHRP_OGMap3D.l
/home/leus/ros/hydro/devel/share/roseus/ros/hrpsys_ros_bridge/msg/OpenHRP_OGMap3D.l
```

I looked at the code.
Maybe `)` of deserialize method is incorrect?
```
 ...
  (:deserialize
   (buf &optional (ptr- 0))
   ;; float64 _resolution                                                                                                                                                                                                                                                                  
     (setq _resolution (sys::peek buf ptr- :double)) (incf ptr- 8)
   ;; hrpsys_ros_bridge/RTC_Point3D _pos                                                                                                                                                                                                                                                   
     (send _pos :deserialize buf ptr-) (incf ptr- (send _pos :serialization-length))
   ;; int16 _nx                                                                                                                                                                                                                                                                            
     (setq _nx (sys::peek buf ptr- :short)) (incf ptr- 2)
   ;; int16 _ny                                                                                                                                                                                                                                                                            
     (setq _ny (sys::peek buf ptr- :short)) (incf ptr- 2)
   ;; int16 _nz                                                                                                                                                                                                                                                                            
     (setq _nz (sys::peek buf ptr- :short)) (incf ptr- 2)
   ;; uint8[] _cells                                                                                                                                                                                                                                                                       
   (let ((n) (sys::peek buf ptr- :integer)) (incf ptr- 4))
     (setq _cells (make-array n :element-type :char))
     (replace _cells buf :start2 ptr-) (incf ptr- n)))
   ;;                                                                                                                                                                                                                                                                                      
   self) ;; parenthesis does not match here??
)
```

k-okada, 2015-04-20:
sorry for that , fixed again

furushchev, 2015-04-21:
@k-okada Sorry for very late notice, is there thing to do anymore?

k-okada, 2015-04-21:
no, i think i've fixed issue.

furushchev, 2015-04-21:
@k-okada OK. Thank you very much :+1:

snozawa, 2015-04-28:
This is resolved in the latest roseus.
(may be fixed through https://github.com/jsk-ros-pkg/geneus/pull/22)
Thanks again!

closed 2015-04-28 by snozawa