jsk-ros-pkg/jsk_roseus #298: roseus crashes? when inputs to stdin pipeline more than 2 times.

furushchev, 2015-05-01:
I wanted to launch roseus in python subprocess, which is hung up when write S-exp to roseus stdin pipeline more than 2 times.

Is there any difference between
- to execute `roseus` and input S-exp to their REPL(normal way):

```bash
$ rosrun roseus roseus
configuring by "/home/furushchev/ros/hydro_parent/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0.0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x5d8e800[16374] --> 0x61f2df0[32748] top=37b2
irtgl irtviewer 
EusLisp 9.11(3b3ef24 1.0.4) for Linux64 created on leus-MacBookPro(Thu Apr 9 14:36:16 JST 2015)
roseus ;; loading roseus("78810fb") on euslisp((9.11 leus-MacBookPro Thu Apr 9 14:36:16 JST 2015 3b3ef24 1.0.4))
eustf roseus_c_util 1.irteusgl$ (+ 1 1)
2 
2.irteusgl$ (+ 2 3)
5
```

- to execute `roseus` with pipelining: 

```bash
$ echo '(+ 1 1)' | rosrun roseus roseus
;; loading roseus("78810fb") on euslisp((9.11 leus-MacBookPro Thu Apr 9 14:36:16 JST 2015 3b3ef24 1.0.4))
2
nil
nil
[ INFO] [1430462291.265474226]: cell* ROSEUS_EXIT(context*, int, cell**)
```

furushchev, 2015-05-01:
I assume that python executes `roseus` as subprocess, and write to its stdin pipeline, then read results from its stdout/stderr pipeline.

k-okada, 2015-05-01:
rosrun is designed to exit when it finish reading from pipeline, also it
does not go into debug mode on error, all of these features are meant to
compatible with roslaunch execution, to exit program whenever someting goes
wrong and respawn to recover the situation. please read roseus.l also
adding document within the code is welcome. that should goes to
http://euslisp-docs.readthedocs.org/en/latest/roseus/roseus/