jsk-ros-pkg/jsk_roseus #310: parallel state machineについて

h-kamada, 2015-05-08:
古田さんに書いていただいた、join の考えによるparallel state machine と、自分が書いたparallel state machine が実装は違うものの実現できることがかなりかぶってしまったように思っていますが、並列化の方法としてはやはりjoin を使ったものの方があるべき姿でしょうか？

同じことが出来るならばjoin 側だけに絞ったほうが良いでしょうか？

k-okada, 2015-05-08:
止揚というコトバが有ります．これのドイツ語は何でしょう（100KP）

wikipediaあたりで勉強してもらったところで，ソフトウェア同じように発展していくのがいいですね．なので，いくつもの実装を見てよりよいものを残していく，というのがいいともいます．で途中で，もしかしたら，あそこでボツにした方法のほうが良かったんじゃない？というときにぱっと戻れるのがよいでしょう．

joinのときにand, orと考えられるけど，それぞれでどう書くのか，というのが見てみたいところでは有ります．

k-okada, 2015-05-08:
ところで，鎌田くんが書いたコードの PR番号はなに？

#309 は古田くんが書いたもののテストコードになっているのかな？

h-kamada, 2015-05-08:
いえ、　#308 は古田さんのtestコードを元に自分の並列state machine 用に書いたものです

h-kamada, 2015-05-08:
ドイツ語ではアウフヘーベン、ヘーゲルの弁証法で用いられた概念ですね！
僕が書いた並列state-machineは #299 にあります。
古田さんの並列state-machine が先にmergeされているので、まずそのコードを自分のものと比較し、上手く自分のものとmerge出来ないか模索してみます。

closed 2015-05-09 by h-kamada