jsk-ros-pkg/jsk_roseus #312: roseus hangs if roseus failed to generate eusdoc

garaemon, 2015-05-10:
roseus hangs if roseus failed to generate eusdoc.

It outputs a lot of error.
```
ERROR th=0 symbol conflict exitexit in (export '(exit))/home/leus/ros/hydro_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 symbol conflict exitexit in (export '(exit))/home/leus/ros/hydro_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 symbol conflict exitexit in (export '(exit))/home/leus/ros/hydro_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 symbol conflict exitexit in (export '(exit))make[2]: *** [pr2-interface.md] 
```

mmurooka, 2015-05-12:
eba4dd41a4ba849a557a0c5ff744e95cdcf106c7

k-okada, 2015-05-12:
Sorry I can't find version starting from eba... Please choose one of them
https://github.com/jsk-ros-pkg/jsk_roseus/commits/master

mmurooka, 2015-05-13:
Last message was written by @garaemon . Please reply.

garaemon, 2015-05-14:
3c8f69e8d1fd5f0c4097a82d7d50512d17fda404

garaemon, 2015-05-14:
I think it only occurs running `catkin build` without package

k-okada, 2015-05-14:
We just confirmed this error on @wkentaro' s PC, we'll working on this.

k-okada, 2015-05-14:
> I think it only occurs running catkin build without package

@wkentaro also mentioned same situation

k-okada, 2015-05-19:
@wkentaro

could you check geneus,euslisp,jskeus,roseus version under ~/ros/hydro and 
~/ros/hydro_parent
also could you check if you installed deb version of 
geneus,euslisp,jskeus,roseus?

wkentaro, 2015-05-19:
~/ros/indigo/src
* geneus ed4da49
* jskeus : I don't install this.
* euslisp : I don't install this.
* roseus : tag: 1.3.5

~/ros/indigo_parent : I don't install this.

ros-indigo-geneus : 2.1.2-0trusty-20150225-0910-+0000
ros-indigo-jskeus : 1.0.4-1trusty-20150329-0051-+0000
ros-indigo-euslisp : 9.11.0-1trusty-20150326-1240-+0000
ros-indigo-roseus : 1.2.7-0trusty-20150329-0057-+0000

和田 健太郎 / Kentaro Wada
http://wkentaro.com <http://www.wkentaro.com>

k-okada, 2015-05-19:
so, you do not have euslisp under ~/ros/indigo_parent?

wkentaro, 2015-05-19:
Yes.

garaemon, 2015-09-17:
solved

closed 2015-09-17 by garaemon