jsk-ros-pkg/jsk_roseus #313: ros::roseus-add-msgs hangs up if message contains 61 constants

garaemon, 2015-05-13:
roseus cannot handle messages which contains over 61 constants.

https://github.com/garaemon/jsk_roseus/blob/msgs-bag/roseus/msg/String.msg

If you use the message above, `ros::roseus-add-msgs "roseus"` hangs up.

garaemon, 2015-05-13:
https://gist.github.com/garaemon/73b9211513b16acad928
generated `roseus/ros/roseus/msg/String.l`

garaemon, 2015-05-13:
`(shadow '*CONST62* (find-package "ROSEUS::STRING"))` does not return

k-okada, 2015-05-13:
humm, simplest test code is -> https://gist.github.com/k-okada/eba5c9f10c63b0abb4bb
i'll look into it

k-okada, 2015-05-13:
We need to expand https://github.com/euslisp/EusLisp/blob/master/lisp/c/eus.h#L143

garaemon, 2015-05-13:
I'd like to increase it to 256 at least

k-okada, 2015-05-13:
it seems there are a way to extend symbol vector 
```
64.eusgl$ make-package "hoge"
#<package #X6829728 hoge>
65.eusgl$ (length ((find-package "hoge") . intsymvector))
60
66.eusgl$ (length ((find-package "GEO") . intsymvector))
1828
67.eusgl$ (length ((find-package "USER") . intsymvector))
163
```

k-okada, 2015-05-13:
```
(dotimes (i 100)
  (eval `(defconstant ,(intern (format nil "*CONST~d*" i)) i))))
(make-package "TEST1")
(dotimes (i 100)
  (eval `(defconstant ,(intern (format nil "TEST1::*CONST~d*" i)) i)))
(make-package "TEST2")
(dotimes (i 100)
  (catch 'error
    (labels ((error (&rest args) (apply #'format *error-output* args)(format *error-output* "~%")(throw 'error nil)))
      (shadow (intern (format nil "*CONST~d*" i)) (find-package "TEST2"))
      (eval `(defconstant ,(intern (format nil "TEST2::*CONST~d*" i)) i)))))
(make-package "TEST3")
(dotimes (i 100)
  (intern (format nil "*CONST~d*" i) (find-package "TEST3"))
  (shadow (intern (format nil "*CONST~d*" i)) (find-package "TEST3"))
  (eval `(defconstant ,(intern (format nil "TEST3::*CONST~d*" i)) i)))
```

k-okada, 2015-05-13:
please try 
```
wget https://gist.githubusercontent.com/k-okada/b37a76caf73fd158cb80/raw/e67a0a2e56dd29f25c513e8e47e8e01ec882a228/gistfile1.txt -O - | bash
```
to fix this problem

k-okada, 2015-05-13:
if you merge and recompile everything, you'll get same result https://github.com/jsk-ros-pkg/geneus/pull/31

k-okada, 2015-05-13:
BTW, which message has such a huge const variable? each message has it'own package (not package, not roseus) and internal symbol table size is defined for each message.

garaemon, 2015-05-13:
this is https://github.com/jsk-ros-pkg/jsk_demos/blob/master/jsk_2015_06_hrp_drc/drc_com_common/msg/OCS2FCSmall.msg

k-okada, 2015-05-13:
I see...

k-okada, 2015-05-13:
I see...

mmurooka, 2015-05-13:
I added 61th constant variable to this message file, and met this error.

I reverted and removed DOOR_DRC_LASVEGAS constant variable in https://github.com/jsk-ros-pkg/jsk_demos/pull/656...

k-okada, 2015-05-13:
you don't have to revert that, please add new variable and try
https://github.com/jsk-ros-pkg/jsk_roseus/issues/313#issuecomment-101499461

if this works, i'll merge https://github.com/jsk-ros-pkg/geneus/pull/31 and if you update geneus, everything should working fine.

k-okada, 2015-05-14:
also fixed in message generation level https://github.com/jsk-ros-pkg/geneus/pull/31

mmurooka, 2015-05-15:
遅くなりましたが確認してなおりました。ありがとうございます。
closeしていただいて大丈夫です。

closed 2015-05-15 by k-okada

mmurooka, 2015-05-15:
必要最小限かは分からないですが、
以下で更新が反映されて直ります。
drc_task_commonを最新にしたPCでは必ず必要です。

```
roscd geneus
git pull
source /opt/ros/hydro/setup.bash
catkin bt --start-with-this
source ~/.bashrc
roscd
rm -rf share/roseus/ros/drc_com_common/
roscd drc_com_common
catkin bt --start-with-this --force-cmake
```