jsk-ros-pkg/jsk_roseus #324: test_parallel_state_machine_sample.launch did not work

k-okada, 2015-06-22:
https://s3.amazonaws.com/archive.travis-ci.org/jobs/67828254/log.txt

```
mapped function: :press-button -> (lambda-closure nil 36877928 0 (&rest args) (send *wash* :press-button))
(:open-door) -> (:put-cloth :put-soap)
(:put-cloth) -> (:close-door)
(:put-soap) -> (:close-door)
(:close-door) -> (:press-button)
(:press-button) -> (:success)
start testing [test-parallel-state-machine]
;; Segmentation Fault.
*** Error in `/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl': double free or corruption (!prev): 0x00000000069b9500 ***
testtest_parallel_state_machine_samples ... FAILURE!
FAILURE: test [test_parallel_state_machine_samples] did not generate test results
  File "/usr/lib/python2.7/unittest/case.py", line 331, in run
    testMethod()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rostest/runner.py", line 160, in fn
    self.assert_(os.path.isfile(test_file), "test [%s] did not generate test results"%test_name)
  File "/usr/lib/python2.7/unittest/case.py", line 424, in assertTrue
    raise self.failureException(msg)
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

[testtest_parallel_state_machine_samples][failed]

```

furushchev, 2015-06-22:
ok, I'll handle it tomorrow morning!

⌘ Yuki Furuta

Assigned #324 to @furushchev.

k-okada, 2015-06-22:
thanks, I'll disable that test for now, please revert if you solved the problem.
https://travis-ci.org/jsk-ros-pkg/jsk_roseus/builds/67857906

furushchev, 2015-06-25:
summary of discuss yesterday:

in `roseus`, lambda is defined like below:

```lisp
1.irteusgl$ #'(lambda (x) x)
(lambda-closure nil 0 0 (x) x)
2.irteusgl$ '(lambda (x) x)
(lambda (x) x)
3.irteusgl$ (let () #'(lambda (x) x))
(lambda-closure nil 0 0 (x) x)
4.irteusgl$ (let (a) #'(lambda (x) x))
(lambda-closure nil 20578296 0 (x) x)
5.irteusgl$ (let (a) '(lambda (x) x))
(lambda (x) x)
```

We think that if `lambda-closure` is allocated in `let`(≒ pointer of bindframe is not 0), after escaping `let` scope it can be released unexpectedly, which causes Segmentation fault.
(cf. https://github.com/jsk-ros-pkg/jsk_pr2eus/issues/33#issuecomment-43348178 )
After removing `#` from `#'(lambda (x) xxx)`, my sample code works well.
This bug is confirmed only in Ubuntu 14.04/ROS Indigo environment.

(BTW, is this related? -> https://github.com/euslisp/EusLisp/issues/71 )

closed 2015-10-20 by k-okada