jsk-ros-pkg/jsk_roseus #331: rosdep install does not work for roseus_smach

yoshimalucky, 2015-07-01:
While building the roseus_smach, I got an error saying actionlib_tutorials is missing. 

Then I tried:
rosdep install roseus_smach

But, it just says "#All required rosdeps installed successfully".
Why doesn't it start to execute apt-get install ros-hydro-actionlib-tutorials?

The source codes are all updated and I know both of run depend and build depend have actionlib_tutorials.

k-okada, 2015-07-01:
please run with -v to see what's happenning there

yoshimalucky, 2015-07-01:
I ran it with -v option:
rosdep install roseus_smach -v

Then, rosdep installed ros-hydro-actionlib-tutorials successfully. 
I think I had to source ~/ros/hydro/devel/setup.bash.

Thank you very much.

closed 2015-07-01 by yoshimalucky