jsk-ros-pkg/jsk_roseus #333: ros::*comple-messages*をtにするとloadできなくなる

garaemon, 2015-07-03:
ros::*compile-messages*をtにするとload package://ができなくなります。

テストコード
```lisp
#!/usr/bin/env roseus
(setq ros::*compile-message* t)
(ros::roseus-add-msgs "sensor_msgs")
(load "package://roseus/euslisp/roseus.l")
```

エラー
```
eustf roseus_c_util compiling file: /home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_roseus/roseus//euslisp/roseus.l
;; loading roseus("86c171f") on euslisp((9.12 fc21 Fri Jun 12 03:48:19 JST 2015 fada0ef 1.0.6))
;; loading roseus("86c171f") on euslisp((9.12 fc21 Fri Jun 12 03:48:19 JST 2015 fada0ef 1.0.6))
; *keyword-package* is assumed to be global
; *compile-message* is assumed to be global
; *compile-message* is assumed to be global
/home/lueda/ros/hydro_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0  0x5a129a0 in (apply #'compile-file srcfile :o obj args)COMPILER:E: 
```

ちなみに、`*compile-messages*`をnilにすれば動く
```lisp
#!/usr/bin/env roseus
(setq ros::*compile-message* t)
(ros::roseus-add-msgs "sensor_msgs")
(setq ros::*compile-message* nil)
(load "package://roseus/euslisp/roseus.l")

```

YoheiKakiuchi, 2015-07-03:
そうですね，最近気がつきました．

```lisp
#!/usr/bin/env roseus
(setq ros::*compile-message* t)
(ros::roseus-add-msgs "sensor_msgs")
(setq ros::*compile-message* nil)
```
roseus-add-msgs-with-compile関数か，roseus-add-msgsに:compile tにするような変更がひつようでしょうか．

k-okada, 2015-07-16:
この問題そのものは
compile-file-if-src-newer-so の引数（この形は多分コンパイルできない）＋途中でファイルを消している（roseus.cppからできるroseus.soが消える）
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/euslisp/roseus.l#L359

が原因におもいます．そもそもcompile-file-if-src-newer.so でファイルを消しているのはなぜでしょうか？

YoheiKakiuchi, 2015-07-16:
packagename/hogehoge.lをコンパイルした時に、
コンパイルしたファイル名を pckagename_hogehoge.so したくて、
hogehoge.l をコンパイルしてできた hogehoge.so を packagename_hogehoge.so にコピーして、
hogehoge.so を消しているようです。

k-okada, 2015-07-16:
とりあえずmvしなくても第二引数でディレクトリを指定できそうです．
また，出力先はメッセージファイルのあるdevel/share/roseus/ros/<pkg>はどうでしょうか．

https://github.com/k-okada/jsk_roseus/commit/a91c18e388aa05559acb478a49c64b36ea9bb1c6

YoheiKakiuchi, 2015-07-17:
share/roseus/ros/ ここだと、debに入れた場合はパーミッションが必要なところにならないでしょうか？

garaemon, 2015-07-17:
パッケージのコンパイル時にコンパイルするという作戦はどうでしょうか？

k-okada, 2015-07-17:
/opt/ros .. にある．msg から CMAKE_PREFIX_PATH/devel/share .. に .l をつくって，その.l
から同じ位置に.soを作るので問題ないと思います．

問題あるとするとjadeで.lが/optにインストールされている場合かと思います．

k-okada, 2015-07-17:
デフォルトで ros::*compile-messages* を t にすればそうなると思います．