jsk-ros-pkg/jsk_roseus #366: actionlib-client-stateのasync版でwhileループから抜けない

h-kamada, 2015-09-15:
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus_smach/src/state-machine-actionlib.l#L87
`(while (ros::ok)`
からループに入るので、
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus_smach/test/async-join-state-machine-sample.l#L48
この行から抜けられなくなり、
`make-async-join-state-machine` 関数が機能しない

furushchev, 2015-09-15:
`(while (ros::ok)`は`if async`のelse節なので、`:async`が`t`になっているなら、
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus_smach/src/state-machine-actionlib.l#L86
で即座に返るはずです。

h-kamada, 2015-09-15:
serverが上がっていないところでつっかかっているだけでした。すいません。

closed 2015-09-15 by h-kamada

k-okada, 2015-09-15:
そういう時にわかりやすい警告メッセージが出るといいんだけど，可能かな．

h-kamada, 2015-09-18:
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/euslisp/actionlib.l#L117
debug メッセージになっているところをinfoにして常に表示が出るようにすると、とりあえずserverを待っていることは表示として出てくると思いますが、この変更で良いでしょうか。

garaemon, 2015-09-18:
ループの中で、思ったより長かったらwarnを出す、とかでもいいかもしれない

h-kamada, 2015-09-18:
```
       (when (= 0 (mod count 50))
         (ros::ros-warn "[~A] action server is not found" name-space)
         )
```
5秒ごとにwarning出すとかでしょうか....?

garaemon, 2015-09-18:
> 5秒ごとにwarning出すとかでしょうか....?

僕はそれがいいと思います。
メッセージ自体は
"Still waiting for ~A"
とかにしましょう。

待ってる時間も出してあげるのが親切かな

h-kamada, 2015-09-18:
PR出しました。
jsk-ros-pkg/jsk_roseus#370