jsk-ros-pkg/jsk_roseus #404: roseus.l というファイルがカレントディレクトリにあるとエラーになる

yurikadoi, 2015-11-18:
```
$ roseus
configuring by "/opt/ros/indigo/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x54af9f0[16374] --> 0x590b5a0[32748] top=3f7f
irtgl irtviewer 
EusLisp 9.15( 1.0.10) for Linux64 created on host02.storm.ros.org(Mon Aug 31 15:37:47 UTC 2015)
/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 no such package "ROS""ROS" in #<compiled-code #X54f0128>E: (exit)
mech-user@test1-pc:~/2015winter/2015-soft3/20151104/src/enshu_20151104$ ls
camera.launch  roseus.l~  sample.launch   sample2.launch
roseus.l       roseus2.l  sample.launch~
```

yurikadoi, 2015-11-18:
https://github.com/euslisp/jskeus/issues/292
と似た問題？

YuOhara, 2015-11-18:
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/euslisp/roseus.l
を読まなければならないのに，カレントディレクトリのroseus.lをよんじゃってるみたいです．

yurikadoi, 2015-11-22:
ご回答有難うございます。
よんでほしいのはカレントディレクトリのroeus.lなのにそれが読まれていないのが問題かと思います。
自分で勝手にかいたファイルの名前をroesus.lにするとまずいのでしょうか

YuOhara, 2015-11-22:
他の言語も見てみると多分良くて、参考までにpythonとかだと、
```
leus@leus-dynabook-T654-68KB:~/tmp$ touch python.py
leus@leus-dynabook-T654-68KB:~/tmp$ python
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```
python.pyというファイルを置いても動いています。（全機能がちゃんと動くかは確認できないですが、多分問題ないでしょう）

YuOhara, 2015-11-22:
@yurikadoi さん。問題と思う理由は、やはり
* 初めてroseusに触れる人が行なってしまいがちであるということ
* エラー文が見にくくて,何がおかしいのかが全くわからない
ということでしょうか？

yurikadoi, 2015-11-23:
@YuOhara さん
お返事ありがとうございます。その通りです。

YuOhara, 2015-11-24:
試しに*load-path*をreverseしても意味がなかったので(roseus/bin/roseusを変更しコンパイル)、
```
ARG_STR=("(progn (pushnew \"${ROSEUS_DIR}/euslisp/\" *load-path* :test #'equal) (setq *load-path* (reverse *load-path*))" \
```
load-pathに関係なく、カレントディレクトリからファイルを探しているように見えます。

PR #515 (2017-06-21): tell full path of roseus diretory when load roseus.l ...
close #404
closed 2017-06-21 by k-okada