jsk-ros-pkg/jsk_roseus #407: subscriberを2回読んだ時の正しい挙動は？

mmurooka, 2015-11-19:
```
(ros::subscriber "/chatter" std_msgs::String #'(lambda (x) (print 1))
(ros::subscriber "/chatter" std_msgs::String #'(lambda (x) (print 2))
```
としたとき，1, 2の両方呼ばれるのが正解でしょうか？あるいは2だけ呼ばれるのが正解でしょうか？

なんとなくですが，今のroseusは2だけ表示されて，rospy, roscpp的なものは両方呼ばれる可能性が高い気がしています．

garaemon, 2015-11-19:
eus的には何度もloadするから2だけの挙動が望ましいと思っています。

furushchev, 2015-11-19:
今は他の言語と違って`ros::subscribe`がオブジェクトを返さないので、これを維持するならば2の挙動のほうがいいと思っています。

複数のcallbackを呼ぶcallbackをlambdaで登録しておくのが現在のやり方でしょうか。

k-okada, 2015-11-19:
なるほど
```
(setq s1
(ros::subscriber "/chatter" std_msgs::String #'(lambda (x) (print 1)))
(setq s2
(ros::subscriber "/chatter" std_msgs::String #'(lambda (x) (print 2)))
```
と
```
(setq s
(ros::subscriber "/chatter" std_msgs::String #'(lambda (x) (print 1)))
(setq s
(ros::subscriber "/chatter" std_msgs::String #'(lambda (x) (print 2)))
```
は同じ挙動になることになる。
まあそうかもしれない

そもそも今日
```
(setq s1
(ros::subscriber "/chatter" std_msgs::String #'(lambda (x) (print 1)))
(let ((s2
(ros::subscriber "/chatter" std_msgs::String #'(lambda (x) (print 2))))
```
として(実際はactionlib-clientだけど)大ハマりしたのが原因なんだけど、2回subscribeしたよ、という表示を目立つようにしたらよいの？？？

k-okada, 2015-11-19:
具体的には
```
(set *c* (instance actionlib-client ....))
(send *c* :wait-for-result)  ; ok
(subscribe ".../status")
(send *c* :wait-for-result)  ; ng
```
となりますよね

k-okada, 2015-11-19:
:wait-for-resultするとき、そのコールバック関数が自分のオブジェクトにむすびついているかみる？

furushchev, 2015-11-19:
statusは定期的に来るものなので、僕は一定時間statusが来なかったらwarnがいいと思います。
そしてstatusをsubscribeはせずに、(send *c* :get-status)をstatusが知りたい時に呼べばいい気がします。

k-okada, 2015-11-19:
うーん
```
(dotimes (i 10000) (ros::subscribe ..)))
```
するとオブジェクトがどんどん作られてGCされない気がする．

％ なぜ今週はこうも基本的なことで躓くのか．．．．

furushchev, 2015-11-19:
```lisp
(setq s
(ros::subscriber "/chatter" std_msgs::String #'(lambda (x) (print 1)))
(setq s
(ros::subscriber "/chatter" std_msgs::String #'(lambda (x) (print 2)))
```

そもそも、これは`s`に`t`がくるだけで、subscribeできたかどうかの確認のみにしか使えないですよね？

subscribeにオプション引数でsubscribeが上書きされた時のcallbackを用意するのはどうでしょうか？