jsk-ros-pkg/jsk_roseus #417: service-call で方が違うときのチェックをして欲しい

k-okada, 2015-11-25:
```
(ros::load-ros-manifest "roseus")
;;;                                                                                                                                        
(defun add-two-ints (req)
  (let ((m (send req :response)))
    (ros::ros-info "Returning [~d + ~d = ~d]"
            (send req :a) (send req :b)
            (+ (send req :a) (send req :b)))
    (send m :sum  (+ (send req :a) (send req :b)))
    ;;m                                                                                                                                    
    ))
;;;                                                                                                                                                                                                                                                                      
(ros::roseus "add_two_ints_server")
(ros::advertise-service "add_two_ints" roseus::AddTwoInts #'add-two-ints)
(do-until-key
```
みたいに間違えると，

```
[ INFO] [1448418192.211156673]: Returning [0 + 7 = 7]
;; Segmentation Fault.
;; in (ros::spin-once)
;; You are still in a signal handler.
;;Try reset or throw to upper level as soon as possible.
;; code=-262554512 x=f059bb40 addr=1727600
```
と原因を言わずにエラーになります．型が違うことぐらいはおしえてくれると嬉しいです．

furushchev, 2017-02-11:
#486 solves this issue. I think it's now ok to close.

k-okada, 2017-02-11:
this is going to be released as 1.6.1

closed 2017-02-11 by k-okada