jsk-ros-pkg/jsk_roseus #429: std_msgs/String を引数に持つsrvファイルでエラーが出ることについての質問

kochigami, 2016-02-19:
std_msgs/String を引数に持つsrvファイルをeuslispで使うとエラーが出てしまい困っています．
何かを勘違いしていると思うのですが，解決策があれば教えていただけませんか．

自分で作ったプログラムの中で，
```
(setq externalCollisionProtection (instance naoqi_bridge_msgs::SetExternalCollisionProtectionEnabledRequest :init))
(send externalCollisionProtection :name "Move")
(send externalCollisionProtection :status ())
(ros::service-call "set_external_collision_protection_status" externalCollisionProtection)
```
のようにサービスを実行しようとして，次のエラーが出ます．

```
/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl roseus-error: cannot find method :serialization-length in (send ros::_name :serialization-length)
```
 つくられた
~/catkin_ws/devel/share/roseus/ros/naoqi_bridge_msgs/srv/SetExternalCollisionProtectionEnabled.lの定義の中は，
```
  (:serialization-length
   ()
   (+
    ;; std_msgs/String _name                                                    
    (send _name :serialization-length)
・・・
```
となっています．
無限ループのようになっていていけないから作り直しが必要なのか，
この形であっていて，使い方がいけないのでしょうか．

例えばnao_interaction_msgs/srv/LearnFace.lの中にも

```
 (:serialization-length
   ()
   (+
    ;; std_msgs/String _name                                                    
    (send _name :serialization-length)...
```
がありました．

```
catkin clean -a 
catkin build -v -i --force-cmake
```
した状態でファイルを作りなおしてもファイル内容は変わらず，同じエラーが出ました．
２台のパソコンで確認しました．

YoheiKakiuchi, 2016-02-19:
```(send externalCollisionProtection :name "Move")```
この部分ですが、nameの定義が、std_msgs/Stringなら
```"Move"```ではなくて、```(instance std_msgs::String :init :data "Move")```を入れるのではないかと思います。

```(send (send externalCollisionProtection :name) :data "Move")``` でもいいかもしれません。

kochigami, 2016-02-19:
エラーが出た理由を理解することができました．明日またやってみます．
ご回答ありがとうございました．

closed 2016-02-19 by kochigami