jsk-ros-pkg/jsk_roseus #470: :look-at メソッドの使い方について

kochigami, 2016-07-19:
[jskeusのマニュアル] (http://euslisp-docs.readthedocs.io/en/latest/jskeus/jmanual/)の中に
```
ロボットの視線を対象物に向けるためのメソッドとして以下のようなものがある．
(send *robot* :head :look-at (send *kettle* :worldpos))
```
とあるのですが、このメソッドの使い方がよく分かりません。

```:look-at``` メソッドはどこに定義されているのでしょうか。
pepper-interface.l, naoqi-interface.l, robot-interface.lには見つかりませんでした。

例えば、ペッパーで以下のように使ってみると、頭の上を物体に向けたような姿勢になってしまいます。

```
roseus pepper-interface.l

(pepper-init t)
(setq *box* (make-cube 100 100 200))
(send *box* :set-color :blue)
(send *box* :locate (float-vector 900 50 900) :world)
(objects (list *pepper* *box*))
(send *pepper* :head :look-at (float-vector 900 50 900))
```
[image]

k-okada, 2016-07-19:
https://github.com/euslisp/jskeus/blob/master/irteus/irtrobot.l#L166
ですね。
```
(send *pepper* :head :look-at (float-vector 900 50 900) :debug-view t)
```
するとheadのend-coordsがちゃんとついていないということで、peppereus/pepper.yamlを見ると
```
## TODO: end-coords tokuni base
```
となっていますね。例えば以下のようにしてpeppereusをコンパイルし直すとうまく行くと思います。
larm/rarm/lleg/rleg?のend-coordsもうまくつけてもらえると嬉しいです。

```
## end-coords
##
head-end-coords:
  parent : CameraTop_frame
  translate : [0, 0, 0]
  rotate    : [0, 1, 0, 90]
```

k-okada, 2016-07-19:
% end-coordsをちゃんと付けずにikをとくのは難しかったかも。。。

kochigami, 2016-07-19:
ご教示ありがとうございました。
元のend-coordsの定義はどこから来ているのかについても調べようと思います。

[image]

分かったこと：
``` :head :look-at ``` は顔のend-coordsに対して解くik
ikがうまく行かない時にend-coordsのことを気にする

pepper.yamlにend-coordsの記述はないが、
``` send *pepper* :head (:rarm, :larm, :rleg) :end-coords ```
をやってみると元々用意されている
しかしこの値がよくない

元々定義にあった ``` send *pepper* :head :end-coords :draw-on :size 100 :flush t ``` などの結果
[image]

```:debug-view t``` のログの見方　（move-target: head-end-coordsの座標が悪いということで判断するらしい）

head-end-coords 変更前
```
send *pepper* :head :look-at (float-vector 900 50 900) :debug-view t
loop:   1
union-link-list: (Neck Head)
move-target: (#<cascaded-coords #X64683c0 :head-end-coords  -38.0 0.0 169.9 / 0.053 0.637 0.0>)
targe-coordst: (#<coordinates #X6ad9948  900.0 50.0 900.0 / -0.033 0.908 -0.068>)
vel-pos :
vel-rot :  0.000   0.273 
vel-pos-norm :   0.000/  1.000
vel-rot-norm :   0.273/  0.017
angle :    3.1    36.5 
 min  : -119.5   -40.5 
 max  :  119.5    36.5 
usrwei:  1.000   1.000 
ocost :  0.024 1102.752 
cost  :  0.025 1102.752 
weight:  0.976   0.001 
k     :  0.000 (manipulability:  0.595, gain:  0.001, limit:  0.100, len:2)
nspace: -0.000  -0.001 
J     : -0.595   0.000 
         0.000   1.000 
d(JJt):  0.354
J#t   : -1.681   0.000 
         0.000   1.000 
x     :  0.000   0.273 
J#x   : -0.006  15.644 
Ny    :  0.000   0.000 
dav   : -0.006  15.644 
dav^  : -0.006  15.644 
loop:   2
union-link-list: (Neck Head)
move-target: (#<cascaded-coords #X64683c0 :head-end-coords  -38.0 0.0 169.9 / 0.053 0.637 -6.939e-18>)
targe-coordst: (#<coordinates #X6b54af8  900.0 50.0 900.0 / -0.033 0.908 -0.068>)
vel-pos :
vel-rot : -0.000   0.273 
vel-pos-norm :   0.000/  1.000
vel-rot-norm :   0.273/  0.017
angle :    3.0    36.5 
 min  : -119.5   -40.5 
 max  :  119.5    36.5 
usrwei:  1.000   1.000 
ocost :  0.025 1102.752 
cost  :  0.025 1102.752 
weight:  1.000   0.001 
k     :  0.000 (manipulability:  0.595, gain:  0.001, limit:  0.100, len:2)
nspace: -0.000  -0.001 
J     : -0.595   0.000 
        -0.000   1.000 
d(JJt):  0.354
J#t   : -1.681  -0.000 
[... 123 lines cut ...]
vel-pos :
vel-rot : -0.000   0.000 
vel-pos-norm :   0.000/  1.000
vel-rot-norm :   0.000/  0.017
angle :    3.1   -30.0 
 min  : -119.5   -40.5 
 max  :  119.5    36.5 
usrwei:  1.000   1.000 
ocost :  0.025   9.689 
cost  :  0.025   9.738 
weight:  0.976   0.093 
k     :  0.000 (manipulability:  0.866, gain:  0.001, limit:  0.100, len:2)
nspace: -0.000   0.028 
J     : -0.866  -0.000 
         0.000   1.000 
d(JJt):  0.750
J#t   : -1.155   0.000 
         0.000   1.000 
x     : -0.000   0.000 
J#x   :  0.000   0.002 
Ny    : -0.000  -0.000 
dav   :  0.000   0.002 
dav^  :  0.000   0.002 
#f(3.16406 -1.9336 -4.30664 84.9902 10.0195 -70.0488 -19.9512 -39.1142 84.9902 -10.0195 70.0488 19.9512 39.1973 3.05126 -29.9891)
```

closed 2016-07-19 by kochigami

k-okada, 2016-07-19:
end-coordsはJSK（jskeus）ローカルルールで
headは目線の方向，前がｚ，xが下かな．
手はペットボトルを掴んだ時の手の中の（ロボットとてきには手を構成するポリゴンの外）の中心．ｚが上，ｘが前
で適当にエイヤと決めてください．

いまのPepperはpepper.yamlにend-coordsの指定が無いので，デフォルト値（リンクのルート）なので気にしなくていいです．