jsk-ros-pkg/jsk_roseus #500: チュートリアル（パブリッシャ/サブスクライバ）についての質問

M-Onitsuka, 2017-02-10:
roseusのチュートリアルでシンプルなパブリッシャを書く項についての質問です．
```
#!/usr/bin/env roseus

(ros::load-ros-manifest "roseus")
(ros::roseus "talker")
(ros::advertise "chatter" std_msgs::string 1)
(ros::rate 10)
(while (ros::ok)
  (setq msg (instance std_msgs::string :init))
  (send msg :data (format nil "hello world ~a" (send (ros::time-now) :sec-nsec)))
  (ros::ros-info "msg [~A]" (send msg :data))
  (ros::publish "chatter" msg)
  (ros::sleep))
(exit)
```
というコードの中で
```
(send msg :data (format nil "hello world ~a" (send (ros::time-now) :sec-nsec))) ;;(1)
```
となっているところをすぐ下の行の
```
(ros::ros-info "msg [~A]" (send msg :data)) ;;(2)
```
を真似して
```
(send msg :data "hello world ~a" (send (ros::time-now) :sec-nsec)) ;;(3)
```
としたところmismatch argumentのエラーが出ました．
(1)で```format nil```によって文字列が```:data```に渡されていて，(3)では余計な式が入ってしまうので，
引数の数が合わないことがエラーの原因だと考えています．
では(2)ではどのように式が評価されているのでしょうか．

knorth55, 2017-02-10:
`ros::ros-info`の処理が知りたいということですか？
それはここらへんを見るといいと思います

https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/roseus.cpp#L1798
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/roseus.cpp#L729-L744

`(send msg :data string)`という形でstringというargumentを1つのみ受け取る仕様です
format関数はstringを返り値として返してくれるので(1)はうまくいきますが(3)はformat関数がないのでうまくはいきません
一方、`ros-info`はrosの関数を内部で呼んでいるだけでformat関数は呼んでいないと思います
あまり質問の意図がわからなかったのですがこれで回答になっていますか？

M-Onitsuka, 2017-02-10:
回答ありがとうございます．
>(send msg :data string)という形でstringというargumentを1つのみ受け取る仕様です
>format関数はstringを返り値として返してくれるので(1)はうまくいきますが(3)はformat関数がないのでうまくはいきません 
ここは理解出来ました．
```ros-info```の部分ですが，質問の意図としては
```
(ros::ros-info (format nil "msg [~A]" (send msg :data))) ;;(2)'
```
としなくても
```
(ros::ros-info "msg [~A]" (send msg :data)) ;;(2)
```
で動くのはなぜだろうということになると思います．言葉が足らなくて申し訳ありません．

(2)において```ros::ros-info```では引数として```"msg[~A]"```と```(send msg :data)```が渡されています．
(2)'のように先に```(format nil "msg [~A]" (send msg :data))```を評価して一つのstringにしてから渡さなくても動くのはどうなっているんだろうと考えました．

knorth55, 2017-02-10:
roscppのROS_INFOも同様の仕様でそれにならったのではないでしょうか？
内部処理を知りたければソースコードを読むといいと思います

M-Onitsuka, 2017-02-10:
わかりました，とりあえずまずはroscppを読んでみます．
回答ありがとうございました！

knorth55, 2017-02-10:
roseusにおけるros::ros-infoの内部処理についてはhttps://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/roseus.cpp#L729-L744だと思います
roscppのloggingとしてhttp://wiki.ros.org/roscpp/Overview/Loggingを読んでみるといいと思います
ただそういう設計の関数なのだと理解して、内部処理は別に気にしなくてもいいと思いますが

closed 2017-11-20 by M-Onitsuka