jsk-ros-pkg/jsk_roseus #535: 計算した値を四捨五入するのをやめられますか

kochigami, 2017-08-06:
pythonだと、
``` 1.50199099766e+18``` のように表示されるのですが、
euslispだと、
``` 1.501991e+18``` のようになります。

やりたいことは、
時間を記録して、ファイルに書き込むことなので、
今の所は計算をせず、```sec```, ```nsec```両方取り出してファイルに書き込もうと思います。

```
4.E2-irteusgl$ setq time (ros::time-now)
#<ros::time #X57bc058 1502011442.658>
5.E2-irteusgl$ send time :sec
1502011442
6.E2-irteusgl$ send time :nsec
658569503
7.E2-irteusgl$ setq answer (+ (* (send time :sec) 1000000000.0) (send time :nsec))
1.502011e+18
8.E2-irteusgl$ answer
1.502011e+18
9.E2-irteusgl$ send time :now
#<ros::time #X57bc058 1502011531.816>

```

k-okada, 2017-08-06:
```
$ (format t "~32,16f~%" (/ (send time :to-nsec) 1000000000.0))
     1502013373.6416673660278320
```
はどうでしょう．あとは、
```
$ setq answer (+ (* (send time :sec) 1000000000) (send time :nsec))
1502013373641667228
$ answer
1502013373641667228
```
というほうほうもあります．
あとは、もとのanswerの計算法だと
```
$ (format t "~f" answer)
1502013373641666560.0nil
$ (format t "~32,12f" answer)
1502013373641666560.000000000000
```
でしょうか．

kochigami, 2017-08-06:
ご教示ありがとうございました。

もう一点、もしよろしければご教示ください。
``` (format t "~32,16f~%" (/ (send a :to-nsec) 1000000000.0)) ``` の値をそのまま別の変数に代入したいのですが、できますでしょうか。
今は、記録をまとめて、string メッセージにして、
トピックに乗せて、それをファイルに書くようにしています・・・

```
22.E4-irteusgl$ (setq time_and_str (format nil "~a, ~a" (format t "~32,16f~%" (/ (send a :to-nsec) 1000000000.0)) "hoge"))
     1502019885.0964641571044922
"nil, hoge"
23.E4-irteusgl$ time_and_str
"nil, hoge"
```

k-okada, 2017-08-06:
$ (setq xxx (format nil "~32,16f" (/ (send time :to-nsec) 1000000000.0)) )
"     1502013373.6416673660278320"
$ (print abc)
"     1502013373.6416673660278320
"
"     1502013373.6416673660278320
"

でどうでしょう．

kochigami, 2017-08-06:
ご教示ありがとうございました。

format t/ nil の違いがいまいち分かっていないので、
勉強します。

k-okada, 2017-08-07:
formant t は標準出力への書き出し
nilは書き出さない
どちらにせよformat関数はフォーマットされた文字を返すので、nilで何も標準させずに、変数に割り当てる
と言うことをしています

kochigami, 2017-08-15:
すみません、今度はファイルに書き込んだ値を、
プログラムで読み込むときにまとめられるのをやめたいのですが、方法が分かりません。

たくさんファイルに書いてある数の大きさを比べたいのですが、
まとめられてしまっていて、全て同じ大きさと出てしまい困っています。

```
setq time 1502154610.7192630767822266
1.502155e+09

(setq time (format nil "~32,16f" time))
"     1502154610.7192630767822266"
read-from-string time
1.502155e+09
```

tmp.l
```
    1502154610.7192630767822266 help_me
```

``` 
(with-open-file (s "tmp.l" :direction :input)(while (setq b (read s nil))(print b)))
1.502155e+09
help_me
nil
```

YoheiKakiuchi, 2017-08-15:
以下のような感じで、プリントXXXe+ZZZと同じように表示されていても、
イコールではなくて、表示桁数を増やせば正しい値が入っているので、
formatで必要な桁数を書きだせば、読み込んだ後で比較できると思います。

~~~
(setq timea 1502154610.7192630767822266)
1.502155e+09
(setq str (format nil "~32,16f" timea))
"     1502154610.7192630767822266"
(setq timea-str (read-from-string str))
1.502155e+09

(setq timeb 1502154610.719264)
1.502155e+09
(setq str (format nil "~32,16f" timeb))
"     1502154610.7192640304565430"
(setq timeb-str (read-from-string str))
1.502155e+09

(= timeb timeb-str)
t
(= timea timea-str)
t
(= timea timeb)
nil
(= timea-str timeb-str)
nil
(- timea timeb)
-9.536743e-07
(- timea-str timeb-str)
-9.536743e-07
~~~