jsk-ros-pkg/jsk_roseus #543: roseusのactionlibのgoalの型について

dyson8910, 2017-11-08:
pythonもCも`FibonacciGoal`型なのに、
https://github.com/ros/common_tutorials/blob/indigo-devel/actionlib_tutorials/src/fibonacci_class_client.cpp#L20
https://github.com/ros/common_tutorials/blob/indigo-devel/actionlib_tutorials/scripts/fibonacci_client.py#L23
なぜ、roseusだけ`FibonacciActionGoal`を要求するのでしょうか？
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus/test/fibonacci-client.l#L33

Affonso-Gui, 2017-11-16:
僕も詳しくないので、間違っていたら誰か訂正してください。

どの場合でもゴールtopicの型は`*ActionGoal`になっていて、メッセージの中身を覗くと
```
affonso@P40:~$ rosmsg show actionlib_tutorials/FibonacciActionGoal 
std_msgs/Header header
  uint32 seq
  time stamp
  string frame_id
actionlib_msgs/GoalID goal_id
  time stamp
  string id
actionlib_tutorials/FibonacciGoal goal
  int32 order
```
のように`header`、`goal_id`、`goal`から構成されています。その中、`goal`が必ず`*Goal`型になっています。

Python や C の`send_goal`がやっているのは、要するに`goal`に`header`と`goal_id`を付けて`*ActionGoal`としてpublishすることです。それに対して、roseus では `*ActionGoal`を引数に取り、そのままpublishしています。

気になるのであれば、例えば以下のようなwrapper関数を定義し
```
(defmethod ros::simple-action-client
  (:send-action-goal (goal &rest args)
    (let ((action-goal (instance (let ((str (format nil "~a" (send (class goal) :name)))) (symbol-value (read-from-string (concatenate string (subseq str 0 (- (length str) 4)) "actiongoal")))) :init)))
      (send action-goal :header :stamp (ros::time-now))
      (send* self :send-goal action-goal args))))
```
Python や C っぽく`*Goal`型で送れます。
```
    (setq goal (instance actionlib_tutorials::FibonacciGoal :init))
    (send goal :order order)
    (send *c* :send-action-goal goal :feedback-cb #'feedback-cb)
```

(あまりお勧めできませんが)

dyson8910, 2017-11-22:
ありがとうございます。
とりあえずはFibonacciActionGoal型でやってみます。
(wrapper関数については時間があればやってみようと思います。)

closed 2017-11-22 by dyson8910

k-okada, 2017-11-23:
うーん．最初に作った時に失敗しましたねぇ．
https://github.com/jsk-ros-pkg/jsk_roseus/pull/546
につくてみました．