jsk-ros-pkg/jsk_roseus #544: std_msgs::MutliArrayLayoutのroseusにおけるinstance生成

M-Onitsuka, 2017-11-15:
自分の中ですでに納得はしているのですが、Auto-generateされた```std_msgs::MutiArrayLayout```のソースを読んでいた際、```:init :dim```で```_dim```に結局何も```push```されていないので生じた疑問です。
roseusにおいて多次元配列を送るために```std_msgs::MultiArrayLayout```クラスのインスタンスを生成する際、
スロット変数であるdimにはデフォルトで```nil```が入ります。
dimを含めてLayoutを、更には```std_msgs::Float64MultiArray```などをpublishしたいとき、```std_msgs::MutiArrayDimension```のインスタンスを自分で別に生成してLayoutのdimに代入する必要があるのでしょうか？
また、自分で```std_msgs::MutiArrayDimension```のインスタンスを渡す際、多次元配列の場合は```std_msgs::MutiArrayDimension```のインスタンスをlistにして渡せばいいのでしょうか?

以下hoge/share/roseus/ros/std_msgs/msg/MultiArrayLayout.lの一部
```
;; Auto-generated. Do not edit!                                                                       

(when (boundp 'std_msgs::MultiArrayLayout)
  (if (not (find-package "STD_MSGS"))
    (make-package "STD_MSGS"))
  (shadow 'MultiArrayLayout (find-package "STD_MSGS")))
(unless (find-package "STD_MSGS::MULTIARRAYLAYOUT")
  (make-package "STD_MSGS::MULTIARRAYLAYOUT"))

(in-package "ROS")
;;//! \htmlinclude MultiArrayLayout.msg.html                                                          

(defclass std_msgs::MultiArrayLayout
  :super ros::object
  :slots (_dim _data_offset ))

(defmethod std_msgs::MultiArrayLayout
  (:init
   (&key
    ((:dim __dim) (let (r) (dotimes (i 0) (push (instance std_msgs::MultiArrayDimension :init) r)) r)) 
;;何もしていない
    ((:data_offset __data_offset) 0)
    )
   (send-super :init)
   (setq _dim __dim)
   (setq _data_offset (round __data_offset))
   self)
  (:dim
   (&rest __dim)
   (if (keywordp (car __dim))
       (send* _dim __dim)
     (progn
       (if __dim (setq _dim (car __dim)))
       _dim)))
;;(以下略)
```

k-okada, 2017-11-16:
確かに気持ち悪いですね
https://github.com/jsk-ros-pkg/geneus/pull/55
で生成すると，
```
(defmethod std_msgs::MultiArrayLayout
  (:init
   (&key
    ((:dim __dim) ())
    ((:data_offset __data_offset) 0)
    )
```
になるので，これだと納得でしょうか．たしかに`:dim 10` としたら，listができてdimに入れてくれてもいい気もしますが，pythonでも
```
In [2]: MultiArrayLayout()
Out[2]: 
dim: []
data_offset: 0

In [3]: :

In [3]: MultiArrayLayout(dim=10)
Out[3]: 
dim: 10
data_offset: 0
```
となるので，良いかな．という気もします．

M-Onitsuka, 2017-11-17:
>これだと納得でしょうか．

そうですね、何もしないという挙動がそのまま書いてあってわかるのでそれだと納得です。
>良いかな．という気もします．

pythonの方のソースも以下を読んでみて、型が(明示的に)ない場合はこれでいいのかなと自分も思います。
http://docs.ros.org/diamondback/api/std_msgs/html/__MultiArrayLayout_8py_source.html

ただ、他のクラスの様にinstanceにkeyをどんどん与えてメンバにアクセスしようとするとdimで止まるので、少し混乱しました。

それと追加の質問で、MultiArrayLayoutにdimを与える際は以下のようにすればいいのでしょうか?
```
(setq layout (instance std_msgs::MultiArrayLayout :init))
(setq dim1 (instance std_msgs::MultiArrayDimesion :init))
(setq dim2 (instance std_msgs::MultiArrayDimesion :init))
(setq dim3 (instance std_msgs::MultiArrayDimesion :init))
(send layout :dim (list dim1 dim2 dim3))
```

Affonso-Gui, 2017-11-17:
```
(setq layout (instance std_msgs::MultiArrayLayout :init :dim (make-list 3 :initial-element (instance std_msgs::MultiArrayDimension :init))))
```
でも書けるよ

M-Onitsuka, 2017-11-17:
@Affonso-Gui 
これだと同じID?のインスタンスが3個のリストが作られてしまって、
dimensionを一ついじると他のdimも変更されてしまってまずそう

```
3.irteusgl$ setq lst (make-list 3 :initial-element (instance std_msgs::MultiArrayDimension :init))
(#<std_msgs::multiarraydimension #X5cbdbf8> #<std_msgs::multiarraydimension #X5cbdbf8> #<std_msgs::multiarraydimension #X5cbdbf8>)
4.irteusgl$ send (car lst) :label "hello"
"hello"
5.irteusgl$ send-all lst :label
("hello" "hello" "hello")
```

Affonso-Gui, 2017-11-17:
その通りでしたね。失礼しました。
正しくは自動生成されたコード見たく
```
(setq layout (instance std_msgs::MultiArrayLayout :init :dim (let (r) (dotimes (i 3 r) (push (instance std_msgs::MultiArrayDimension :init) r)))))
```
などとすれば良いはずです。

M-Onitsuka, 2017-11-17:
@Affonso-Gui 
@k-okada 
jsk-ros-pkg/geneus#55
や
先のコメントを参考に
```
 (setq layout (instance std_msgs::MultiArrayLayout :init :dim (let (r) (dotimes (i 10) (push (instance std_msgs::MultiArrayDimension :init) r)) r)))a
```
や
```
(setq arr (instance std_msgs::Float64MultiArray :init))
(send arr :layout :dim (let (r) (dotimes (i 3) (push (instance std_msgs::MultiArrayDimension :init) r)) r))
```
とすればよいということで、このissueを閉じたいと思います

closed 2017-11-17 by M-Onitsuka