jsk-ros-pkg/jsk_roseus #554: manifest.l isn't created in workspace when deb is installed

pazeshun, 2018-01-06:
すでにaptで入れてあるパッケージをソースからビルドして使おうとした際、`(ros::load-ros-manifest "package_name")`で読み込まれるパッケージ群が、aptで入れてあるバージョンが依存しているものになり、新しく必要になったパッケージがloadされませんでした。
調べて見ると、`WORKSPACE_PATH/devel/share/roseus/ros`以下に該当パッケージのmanifest.lが無いことに気づきました。
依存関係が更新されるような変更があった場合、releaseまでその変更をソースから使おうとした際に問題になると思うのですが、どうでしょうか。

--------------------------
以下、再現方法です。
baxtereusが以下のようにインストール済みのものとします。
```
$ dpkg -l ros-indigo-baxtereus
要望=(U)不明/(I)インストール/(R)削除/(P)完全削除/(H)保持
| 状態=(N)無/(I)インストール済/(C)設定/(U)展開/(F)設定失敗/(H)半インストール/(W)トリガ待ち/(T)トリガ保留
|/ エラー?=(空欄)無/(R)要再インストール (状態,エラーの大文字=異常)
||/ 名前                                      バージョン                アーキテクチャ            説明
+++-=========================================-=========================-=========================-========================================================================================
ii  ros-indigo-baxtereus                      1.1.0-1trusty-20171225-16 amd64                     The baxtereus package
```
この状態で、
```
$ cd ~
$ mkdir -p test_ws/src
$ cd test_ws/src/
$ git clone https://github.com/jsk-ros-pkg/jsk_robot.git
$ cd ..
$ catkin build
$ cd devel/share/roseus/ros/baxtereus  # manifest.lはこの中にあるはず
bash: cd: devel/share/roseus/ros/baxtereus: そのようなファイルやディレクトリはありません
$ source ~/test_ws/devel/setup.bash
$ roscd baxtereus
$ roseus baxter-interface.l 
configuring by "/opt/ros/indigo/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x5e44ab0[16374] --> 0x62a36f0[32748] top=3d1c
irtgl irtglc irtviewer 
EusLisp 9.23( 1.1.0) for Linux64 created on ip-172-31-23-158(Mon Mar 13 06:38:31 PDT 2017)
roseus ;; loading roseus("1.6.3") on euslisp((9.23 ip-172-31-23-158 Mon Mar 13 06:38:31 PDT 2017  1.1.0))
eustf roseus_c_util 
;; extending gcstack 0x62a36f0[32738] --> 0x6dc4720[65476] top=7cc2
Call Stack (max depth: 20):
/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 no such package "BAXTER_CORE_MSGS""BAXTER_CORE_MSGS" in #<compiled-code #X5efd1d0>E: 
```
となり、baxter_core_msgsがloadされません。
`ros::load-ros-manifest`でロードされている`/opt/ros/indigo/share/roseus/ros/baxtereus/manifest.l`を見ると、
```
(ros::load-ros-package "actionlib")
(ros::load-ros-package "actionlib_msgs")
(ros::load-ros-package "actionlib_tutorials")
(ros::load-ros-package "audio_common_msgs")
(ros::load-ros-package "control_msgs")
(ros::load-ros-package "diagnostic_msgs")
(ros::load-ros-package "dynamic_reconfigure")
(ros::load-ros-package "geometry_msgs")
(ros::load-ros-package "move_base_msgs")
(ros::load-ros-package "nav_msgs")
(ros::load-ros-package "pr2_controllers_msgs")
(ros::load-ros-package "pr2_mechanism_msgs")
(ros::load-ros-package "pr2_msgs")
(ros::load-ros-package "roscpp")
(ros::load-ros-package "roseus")
(ros::load-ros-package "rosgraph_msgs")
(ros::load-ros-package "sensor_msgs")
(ros::load-ros-package "shape_msgs")
(ros::load-ros-package "sound_play")
(ros::load-ros-package "std_msgs")
(ros::load-ros-package "std_srvs")
(ros::load-ros-package "tf")
(ros::load-ros-package "tf2_msgs")
(ros::load-ros-package "topic_tools")
(ros::load-ros-package "trajectory_msgs")
(ros::load-ros-package "visualization_msgs")
(ros::load-ros-package "baxtereus")
```
となり、baxter_core_msgsがありません。
`sudo apt-get purge ros-indigo-baxtereus`をした後、もう一度`catkin build`すると、`~/test_ws/devel/share/roseus/ros/baxtereus/manifest.l`が生成されます。中身をみると、
```
;;
;; DO NOT EDIT THIS FILE
;;
;; THIS FILE IS AUTOMATICALLY GENERATED
;;  FROM /home/pazeshun/test_ws/src/jsk_robot/jsk_baxter_robot/baxtereus/./package.xml (1.1.0)
;; USING /opt/ros/indigo/lib/python2.7/dist-packages/geneus/geneus_main.pyc /opt/ros/indigo/share/geneus/package.xml (2.2.6)
;;
(ros::load-ros-package "actionlib")
(ros::load-ros-package "actionlib_msgs")
(ros::load-ros-package "actionlib_tutorials")
(ros::load-ros-package "audio_common_msgs")
(ros::load-ros-package "baxter_core_msgs")
(ros::load-ros-package "baxter_maintenance_msgs")
(ros::load-ros-package "bond")
(ros::load-ros-package "control_msgs")
(ros::load-ros-package "diagnostic_msgs")
(ros::load-ros-package "dynamic_reconfigure")
(ros::load-ros-package "dynamic_tf_publisher")
(ros::load-ros-package "geometry_msgs")
(ros::load-ros-package "jsk_footstep_msgs")
(ros::load-ros-package "jsk_recognition_msgs")
(ros::load-ros-package "move_base_msgs")
(ros::load-ros-package "moveit_msgs")
(ros::load-ros-package "nav_msgs")
(ros::load-ros-package "nodelet")
(ros::load-ros-package "object_recognition_msgs")
(ros::load-ros-package "octomap_msgs")
(ros::load-ros-package "pcl_msgs")
(ros::load-ros-package "posedetection_msgs")
(ros::load-ros-package "pr2_controllers_msgs")
(ros::load-ros-package "pr2_mechanism_msgs")
(ros::load-ros-package "pr2_msgs")
(ros::load-ros-package "roscpp")
(ros::load-ros-package "roseus")
(ros::load-ros-package "rosgraph_msgs")
(ros::load-ros-package "sensor_msgs")
(ros::load-ros-package "shape_msgs")
(ros::load-ros-package "sound_play")
(ros::load-ros-package "std_msgs")
(ros::load-ros-package "std_srvs")
(ros::load-ros-package "tf")
(ros::load-ros-package "tf2_msgs")
(ros::load-ros-package "topic_tools")
(ros::load-ros-package "trajectory_msgs")
(ros::load-ros-package "visualization_msgs")
(ros::load-ros-package "baxtereus")
```
となっており、baxter_core_msgsが入っています。この状態なら、
```
$ roscd baxtereus/
$ roseus baxter-interface.l 
configuring by "/opt/ros/indigo/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x6194ab0[16374] --> 0x65f36f0[32748] top=3d1c
irtgl irtglc irtviewer 
EusLisp 9.23( 1.1.0) for Linux64 created on ip-172-31-23-158(Mon Mar 13 06:38:31 PDT 2017)
roseus ;; loading roseus("1.6.3") on euslisp((9.23 ip-172-31-23-158 Mon Mar 13 06:38:31 PDT 2017  1.1.0))
eustf roseus_c_util 
;; extending gcstack 0x65f36f0[32738] --> 0x71019f0[65476] top=7cc2
1.irteusgl$ 
```
となり、正常にloadされました。
これは、https://github.com/jsk-ros-pkg/jsk_robot/pull/779で追加された依存関係がまだreleaseされていないので、https://github.com/jsk-ros-pkg/jsk_robot/pull/871のように`ros::load-ros-manifest`だけしかないと、apt側の依存関係が足りないmanifest.lがloadされてしまうのだと思います。

k-okada, 2018-01-06:
なるほど． #555 で治った気がします．
https://github.com/jsk-ros-pkg/jsk_robot/issues/823 にあるように，msg/srv がないパッケージは将来的(jadeから）debでも，manifest.l が作られなくなるので，必要なメッセージの `(load-ros-{package,msg,srv})` に置き換えておくことをおすすめします．これらは，それが依存するファイルも読み込んでくれるはずなので，以下のようにbaxter-interface.l のなかで使われているメッセージのものに変えておけばいいとおもいます．

```
-(ros::load-ros-manifest "baxtereus")
+(ros::load-ros-package "baxter_core_msgs")
```

furushchev, 2018-01-06:
@k-okada 横から失礼します。参考までに質問があります。

> msg/srv がないパッケージは将来的(jadeから）debでも，manifest.l が作られなくなるので，必要なメッセージの (load-ros-{package,msg,srv}) に置き換えておくことをおすすめします．

これは`ros::load-ros-manifest`が使えなくなることを意味しているでしょうか？
その場合、`ros::load-manifest`を`ros::load-ros-package`にエイリアスするような感じにはできなさそうでしょうか？

k-okada, 2018-01-06:
いや使えます．そもそもmanifest.l はrospack depends したものをすべて並べて，load-ros-packagesをしていますが，
msg.l でもそこで使うmsgをroseus-add-msgs
しているので，そもそもmanifest.lは必要なかったですね，ということになっていますが，
jade/kineticでもmsgがあれば，manifest.l はあるので，load-ros-manifestはそのままにしておきます．

msg/srvがあるパッケージをload-ros-manifest しているコードはそのままで動きますが，
msg/srvがないと動かなくなるので，そこは変更しておいてください．というこで，どうせ変更するなら，load-ros-packageにしておいたほうがいいですね．ということでした．

furushchev, 2018-01-06:
@k-okada なるほど、理解しました。ありがとうございます。

pazeshun, 2018-01-06:
@k-okada 
私の理解が正しければ、多くのプログラムがjadeから動かなくなっている気がするのですが、気のせいでしょうか。
例えば、jsk_robotをgit cloneしてきて、その中で`grep -r "ros::load-ros-manifest"`とすると、以下のように出てきます。
```
jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_image_transport/roi-reconfigure-call.l:(ros::load-ros-manifest "dynamic_reconfigure")
jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_image_transport/roi-reconfigure-call.l:(ros::load-ros-manifest "geometry_msgs")
jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_image_transport/roi-reconfigure-call.l:(ros::load-ros-manifest "jsk_topic_tools")
jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_move_base/check-movebase-feedback.l:(ros::load-ros-manifest "pr2eus")
jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_lifelog/pr2-tweet-log.l:(ros::load-ros-manifest "jsk_demo_common")
jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_lifelog/visualization/visualize_objectdetection.l:(ros::load-ros-manifest "posedetection_msgs")
jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_lifelog/visualization/visualize_objectdetection.l:(ros::load-ros-manifest "geometry_msgs")
jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_rocon/cameraman_sample/pr1012_tracker_with_hand.l:(ros::load-ros-manifest "geometry_msgs")
jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_rocon/cameraman_sample/pr1012_tracker_with_base.l:(ros::load-ros-manifest "geometry_msgs")
jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_rocon/look_sample/pr1012_pose_publisher.l:(ros::load-ros-manifest "geometry_msgs")
jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_rocon/look_sample/pr1012_pose_subscriber.l:(ros::load-ros-manifest "geometry_msgs")
jsk_naoqi_robot/jsk_pepper_startup/sample/sample.l:(ros::load-ros-manifest "jsk_pepper_startup")
jsk_naoqi_robot/jsk_pepper_startup/nodes/tweet.l:(ros::load-ros-manifest "roseus")
jsk_naoqi_robot/jsk_pepper_startup/nodes/joy-client.l:(ros::load-ros-manifest "peppereus")
jsk_naoqi_robot/naoeus/euslisp/test_nao_controller.l:(ros::load-ros-manifest "roseus")
jsk_naoqi_robot/naoeus/euslisp/test_nao_controller.l:(ros::load-ros-manifest "actionlib")
jsk_naoqi_robot/naoeus/euslisp/test_nao_walker.l:(ros::load-ros-manifest "roseus")
jsk_naoqi_robot/naoeus/euslisp/nao-interface.l:(ros::load-ros-manifest "naoeus")
jsk_naoqi_robot/naoeus/euslisp/nao_mirror.l:(ros::load-ros-manifest "naoeus")
jsk_naoqi_robot/naoqieus/naoqi-interface.l:(ros::load-ros-manifest "naoqieus")
jsk_naoqi_robot/jsk_201504_miraikan/euslisp/mirai-demo-20150413.l:(ros::load-ros-manifest "peppereus")
jsk_baxter_robot/baxtereus/baxter-interface.l:(ros::load-ros-manifest "baxtereus")
jsk_baxter_robot/baxtereus/test/test-baxter-moveit.l:(ros::load-ros-manifest "baxtereus")
jsk_robot_common/jsk_robot_startup/util/initialpose_publisher.l:(ros::load-ros-manifest "geometry_msgs")
jsk_robot_common/jsk_robot_startup/util/initialpose_publisher.l:(ros::load-ros-manifest "topic_tools")
jsk_robot_common/jsk_robot_startup/lifelog/tweet_client.l:(ros::load-ros-manifest "roseus")
jsk_robot_common/jsk_robot_startup/lifelog/tweet_client.l:(ros::load-ros-manifest "topic_tools")
jsk_robot_common/jsk_robot_startup/lifelog/time-signal.l:(ros::load-ros-manifest "roseus")
jsk_robot_common/jsk_robot_startup/lifelog/tweet_client_warning.l:(ros::load-ros-manifest "diagnostic_msgs")
jsk_robot_common/jsk_robot_startup/lifelog/speaking-program-is-started-or-terminated.l:(ros::load-ros-manifest "roseus")
jsk_robot_common/jsk_robot_startup/lifelog/active_user.l:(ros::load-ros-manifest "roseus")
jsk_robot_common/jsk_robot_startup/lifelog/active_user.l:(ros::load-ros-manifest "pr2_mechanism_controllers")
jsk_robot_common/jsk_robot_startup/lifelog/active_user.l:(ros::load-ros-manifest "sensor_msgs")
jsk_robot_common/jsk_robot_startup/lifelog/active_user.l:(ros::load-ros-manifest "diagnostic_msgs")
jsk_robot_common/jsk_robot_startup/lifelog/tweet.l:(ros::load-ros-manifest "roseus")
jsk_robot_common/jsk_robot_startup/lifelog/test/test-logger.l:(ros::load-ros-manifest "geometry_msgs")
jsk_robot_common/jsk_robot_startup/lifelog/test/test-logger.l:(ros::load-ros-manifest "pr2_controllers_msgs")
jsk_robot_common/jsk_robot_startup/lifelog/test/test-logger.l:(ros::load-ros-manifest "posedetection_msgs")
jsk_robot_common/roseus_remote/euslisp/remote-roseus-client.l:(ros::load-ros-manifest "roseus_remote")
jsk_robot_common/roseus_remote/euslisp/remote-roseus-server.l:(ros::load-ros-manifest "roseus_remote")
jsk_fetch_robot/fetcheus/fetch-interface.l:(ros::load-ros-manifest "fetcheus")
```
これを眺めてみると、msg/srvがないpr2eus, jsk_pepper_startup, peppereus, naoeus, naoqieus, baxtereus, fetcheusをload-ros-manifestしているものは、全て動かなくなる恐れがあると思いました。

knorth55, 2018-01-07:
個人的な理解だと、全て依存関係を`package.xml`に書いておいて自身のpackageの`manifest.l`のみを読む方式になっていると考えていました。
https://github.com/jsk-ros-pkg/jsk_robot/pull/871 はそういう意図でした。
pythonのように使うものは全て文頭でloadするということですか？

k-okada, 2018-01-07:
> msg/srvがないpr2eus, jsk_pepper_startup, peppereus, naoeus, naoqieus, baxtereus, fetcheusをload-ros-manifestしているものは、全て動かなくなる

全ては動かないくなるのではなくて，https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_robot_common/jsk_robot_startup/lifelog/tweet_client.l#L3
```
(ros::load-ros-manifest "roseus")
(ros::load-ros-manifest "topic_tools")
```
みたいにたまたま必要なパッケージをloadしていれば問題ないはずです．
が，実際には問題ありそうですね．https://github.com/jsk-ros-pkg/jsk_robot/issues/823#issuecomment-324808685 のケースも実際にはダメそう．

ということで，
https://github.com/jsk-ros-pkg/jsk_roseus/pull/555/commits/5ec2e6e7609fcd586a3edb7c06716aed0cd983b0
を作ってみました．ただこれは，
https://github.com/jsk-ros-pkg/jsk_roseus/pull/555/commits/87de0da7842afea20598a18d48bbd982ec5c0916
に依存していて，このコードはだいぶ怪しいですね．うまくlibpythonがロードできない時がある気がします．

k-okada, 2018-01-07:
> robot/blob/master/jsk_robot_common/jsk_robot_startup/
> lifelog/tweet_client.l#L3
> が，実際には問題ありそうですね．jsk-ros-pkg/jsk_robot#823 (comment)
> 5ec2e6e
> 87de0da

>
個人的な理解だと、全て依存関係をpackage.xmlに書いておいて自身のpackageのmanifest.lのみを読む方式になっていると考えていました。

現状はそうなんですが，jade/kinetic
からroseusメッセージがデフォルトでコンパイルされてdebに入るようになっているのですが，このコンパイルされるのはmsg/srvがあるパッケージなので

> pythonのように使うものは全て文頭でloadするということですか？

という方式にせざるをえないですね，という話です．

knorth55, 2018-01-07:
なるほど
そうなるとbaxterで使ってるものやAPC系を変更しないとダメですね

PR #555 (2018-07-04): do not load manifest.l when we have source tree, but does not have manifest.l because of missing msg/srv
Closes #554 

package does not have msg/srv does not have manifest.l under `share/roseus/ros/<pkg>` but exists `share/<pkg>`,
previous logic skips this situation and load manifest.l from `/opt/ros/indigo/share/roseus/ros/<pkg>`

Fixed log to check if we have `share/<pkg>` but does not have `share/roseus/ros/<pkg>`, returns `:no-msg-packages`

c.f. https://github.com/jsk-ros-pkg/jsk_robot/issues/823
closed 2018-07-04 by k-okada