jsk-ros-pkg/jsk_roseus #579: Segmentation fault from subscriber defined in dotimes

pazeshun, 2018-09-20:
I want to define multiple subscribers by `dotimes`, but that trial failed with segmentation fault.
Here is the simple sample code:
```
(ros::roseus "test" :anonymous nil)
(ros::roseus-add-msgs "std_msgs")

(setq test-list (list 1 2))
(setq topics (list "~test1" "~test2"))
(dotimes (i (length test-list))
  (ros::subscribe (elt topics i) std_msgs::Bool
                  #'(lambda (a msg) (print (elt test-list a))) i 1))
(ros::spin)
```
When I published to `/test/test1` or `/test/test2`, segmentation fault occurs.
Debugging with @Affonso-Gui found the following code also occurs segmentation fault:
```
(ros::roseus "test" :anonymous nil)
(ros::roseus-add-msgs "std_msgs")

(setq var 0)
(let (b)
  (ros::subscribe "~test" std_msgs::Bool
                  #'(lambda (msg) (print var))))
(ros::spin)
```
When we erase local variable `b` or don't use global variable `var`, segmentation fault doesn't occur.

pazeshun, 2018-09-20:
We found `dotimes` problem comes from `let` by expanding macro:
```
4.irteusgl$ macroexpand '(dotimes (i 1) (print 1))
(let ((i 0) (#:dotimes355 1)) (declare (integer i #:dotimes355)) (while (< i #:dotimes355) (print 1) (setq i (1+ i))) nil)
5.irteusgl$ pprint *
(let
   ((i 0) (#:dotimes355 1))
   (declare (integer i #:dotimes355))
   (while (< i #:dotimes355) (print 1) (setq i (1+ i)))
   nil)
nil
```

furushchev, 2018-09-20:
It looks like this is related to some of the existing issues:
https://github.com/jsk-ros-pkg/jsk_roseus/issues/251
https://github.com/jsk-ros-pkg/jsk_roseus/issues/324

This problem is originated from the limitation of euslisp: a closure is unsupported.

As seen in the link below, `lambda` function in `let` scope is defined differently from it in global scope.
https://github.com/jsk-ros-pkg/jsk_roseus/issues/324#issuecomment-115060420

In the example you mentioned, this problem can be avoided by changing codes like below for example:

```lisp
(ros::roseus "test" :anonymous nil)

(setq test-list (list 1 2))
(setq topics (list "~test1" "~test2"))
(dotimes (i (length test-list))
  (ros::subscribe (elt topics i) std_msgs::Bool
    `(lambda-closure () 0 0 (a msg) (print (elt test-list a))) i 1))
(ros::spin)
```

k-okada, 2018-09-20:
related to https://github.com/euslisp/EusLisp/issues/147#issuecomment-153003343 ?

pazeshun, 2018-09-20:
> related to [euslisp/EusLisp#147 (comment)](https://github.com/euslisp/EusLisp/issues/147#issuecomment-153003343) ?

I think so. This issue comes from the known issue.
The main point of this issue becomes an alert of implicit `let` inclusion like `dotimes`.