jsk-ros-pkg/jsk_roseus #611: Undeclared jsk_pcl_ros, openni_launch dependencies in roseus_tutorials

andersfischernielsen, 2019-04-02:
I'm currently doing some research on static analysis of ROS packages at the IT University of Copenhagen, more specifically dependency checking of `.launch` files in ROS packages.

There seems to be a `$(find jsk_pcl_ros)` and a `$(find openni_launch)` in

- `jsk_roseus/roseus_tutorials/launch/hsi_color_filter.launch`
that's not declared in `jsk_roseus/roseus_tutorials/package.xml`.

This would make `roslaunch` fail on this file, right?

I'm unsure whether this is a dependency bug since I don't know what the role of the `.launch` file is. If it's just a "demo type" file to show what the package is capable of, then it could be classified as being not a bug, but I'm unsure.

YutoUchimi, 2019-04-06:
Thank you for reporting this issue!
I fixed this dependency bug in #613 .

k-okada, 2019-04-07:
@andersfischernielsen thanks for the report, but we do not consider a bug. These codes are example/demo code and we are aware of it.

closed 2019-04-07 by k-okada