jsk-ros-pkg/jsk_roseus #624: ros::timeとros::duration間の演算結果の型について

MechHolmes, 2019-11-19:
rospyにおいて、
time - time = duration
という型を越えた演算結果の実装がなされていますが、roseusにおいては
time - time = time
という結果が返ってきます。
加算においてはtimeに変換されるのに、減算においてdurationにならないのはtime-がtimeクラスのメンバだからでしょうか。rospyと型が違うのでご報告させていただきます。

以下私の環境での実行結果です。

$roseus
1.irteusgl$ (ros::roseus "test")
t
2.irteusgl$ (setq time1 (ros::time-now))
#<ros::time #X55614a3b8988 1574130433.413>
3.irteusgl$ (setq time2 (ros::time 1000000000))
#<ros::time #X55614a38d3c8 1000000000.000>
4.irteusgl$ (ros::time- time1 time2)
#<ros::time #X55614a3edcb8 574130433.413>

pazeshun, 2019-11-19:
roseusにはduration型というものが存在せず、すべてtime型での実装となっています。
https://euslisp-docs.readthedocs.io/en/latest/roseus/roseus/

MechHolmes, 2019-11-19:
1.irteusgl$ (ros::roseus "test")
t
2.irteusgl$ (setq dur (instance ros::duration :init 10))
#<ros::duration #X5644d0700950 10.000>

このようにros::durationのインスタンスが生成でき、

3.irteusgl$ (setq time (ros::time-now))
#<ros::time #X55ae16403eb8 1574131736.473>
4.irteusgl$ (ros::time+ time dur)
#<ros::time #X55ae163cb588 1574131746.473>
5.irteusgl$ (ros::time+ dur time)
#<ros::time #X55ae163fda20 1574131746.473>
6.irteusgl$ (ros::time- time dur)
#<ros::time #X55ae164297c0 1574131726.473>

とros::timeとros::durationの間でも計算できています。また、

7.rteusgl$ (send dur :methods)
(:init :sleep :init :sec :nsec :sec-nsec :now :to-sec :to-nsec :from-sec :from-nsec :prin1 :init :md5sum- :datatype- :connection-header :plist :get :put :name :remprop :prin1 :prin1 :warning :error :slots :methods :super :get-val :set-val)
8.irteusgl$ (send time :methods)
(:init :sec :nsec :sec-nsec :now :to-sec :to-nsec :from-sec :from-nsec :prin1 :init :md5sum- :datatype- :connection-header :plist :get :put :name :remprop :prin1 :prin1 :warning :error :slots :methods :super :get-val :set-val)

とどちらもクラスになっており、:sleepを除くと違いがないように思います。実用上は全く問題ないと思いますが、この2クラスの存在意義がよくわかりません。(仮にtimeのみの実装とするならばtimeに:sleepを実装して統一するような感じでしょうか)

pazeshun, 2019-11-19:
なるほど、失礼しました。
ros::durationは #459 で追加されたものでして、もっぱらsleepを使うためだけに追加された、ということになっています。
私自身は、普段はros::durationを明示的に使うことはなく、ros::duration-sleepを使っていました。

roseusではros::timeしか使わない、という状況が固定化されてから追加されたものになるので、timeの引き算の結果をdurationにするという変更を加えてしまうと、既存のコードが動かなくなるかもしれず、そのままにしてある、といったところでしょうか。

pazeshun, 2019-11-19:
ros::timeの引き算をした結果分だけスリープするというのを一番簡単にしようとすると、以下になりますかね。
```
1.irteusgl$ ros::roseus "test"
t
2.irteusgl$ setq t1 (ros::time-now)
#<ros::time #X63de710 1574137174.992>
3.irteusgl$ setq t2 (ros::time-now)
#<ros::time #X6558340 1574137182.218>
9.irteusgl$ ros::duration-sleep (send (ros::time- t2 t1) :to-sec)
t
```