jsk-ros-pkg/jsk_roseus #626: ros::time-now を呼ぶとERRORになる

k-okada, 2019-11-27:
```
$ cat hoge.l 
(ros::time-now)
```
というプログラムを書いて，
```
$ roseus hoge.l
```
とすると，
```
eustf roseus_c_util Call Stack (max depth: 20):
  0: at (ros::time-now-raw)
  1: at (setq ros::sec-nsec (ros::time-now-raw))
  2: at (send (instance ros::time :init) :now)
  3: at (ros::time-now)
/opt/ros/melodic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0  0x7f103bef7f00 in (ros::time-now-raw)E: 
```
とエラーになります．

これは，以下のように`(load)`で実行したときは少し親切なメッセージがでます．
```
$ roseus
l$ (load "hoge.l")
Call Stack (max depth: 20):
  0: at (ros::time-now-raw)
  1: at (setq ros::sec-nsec (ros::time-now-raw))
  2: at (send (instance ros::time :init) :now)
  3: at (ros::time-now)
  4: at (apply #'ros::load-org-for-ros ros::fullname args)
  5: at (apply #'ros::load-org-for-ros ros::fullname args)
  6: at (let ((ros::fullname fname)) (when (substringp "package://" fname) (setq ros::fullname (ros::resolve-ros-path fname)) (when ros::*compile-message* (let* ((ros::urlname (url-pathname fname)) (package-name (send ros::urlname :host)) (ros::path-name (format nil "~A_~A" package-name (substitute 95 47 (concatenate string (subseq (send ros::urlname :directory-string) 1) (send ros::urlname :name))))) (ros::old-module (find ros::path-name *loaded-modules* :key #'lisp::load-module-file-name :test #'equal))) (unless ros::old-module (let* ((ros::ppath (unix:getenv "CMAKE_PREFIX_PATH")) (dir (format nil "~A/share/roseus/ros/~A" (subseq ros::ppath 0 (position 58 ros::ppath)) package-name))) (unless (probe-file dir) (unix:mkdir dir)) (compiler:compile-file-if-src-newer ros::fullname (format nil "~A/~A" dir ros::path-name)) (return-from load (load (format nil "~A/~A.so" dir ros::path-name) :entry (format nil "___~A" ros::path-name)))))))) (if (null ros::fullname) (error "file ~s not found" fname)) (apply #'ros::load-org-for-ros ros::fullname args))
  7: at (load "hoge.l")
  8: at #<compiled-code #X55ee7e54d3c8>
/opt/ros/melodic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error:  You must call (ros::roseus "name") before creating the first NodeHandle in (ros::time-now-raw)
```

k-okada, 2019-11-27:
ちなみにエラーがでないようにするには
```
$ cat hoge.l
(ros::roseus "test-example")
(ros::time-now)
```
と，`(ros::roseus)`が必要です．

Affonso-Gui, 2019-11-27:
The usual error message is only setted in the reploop function, so errors that happen while loading files from command line go to the default c level error handler.
https://github.com/euslisp/EusLisp/blob/master/lisp/l/toplevel.l#L251
https://github.com/euslisp/EusLisp/blob/master/lisp/c/eus.c#L382-L402

Regarding why the message is not printed correctly in the default error handler, that happens because the default error handler opts to print the pointer instead of the message itself on E_USER cases (?). 
Made a PR for that in https://github.com/euslisp/EusLisp/pull/404

Affonso-Gui, 2019-11-27:
On related topics, you also cannot `reset` errors derived from files loaded on command line because eustop does not give a catcher for that.

Affonso-Gui, 2019-12-06:
@k-okada Does that answers your question?