jsk-ros-pkg/jsk_roseus #644: eusのオブジェクトからpointcloudを作りたい，あるいはpointcloudからeusのモデルを作りたい．

k-okada, 2020-05-13:
https://gist.github.com/k-okada/63849c1f6a734e2e30188614324a439d にサンプルを置きましたが，（２つのプログラムが一つになているのでそのままでは動かない）
`body-to-triangles` が結構曲者で引数を変えないと無限ループになる場合がある気がします．
 https://github.com/jsk-ros-pkg/euslib/blob/master/jsk/sample_poisson.l でもう少しマシなコードがありますが，https://github.com/euslisp/jskeus/pull/225 を使えるように変更するか，euslib/jsk の
kdtreeを使う必要があるので人手間です．

逆のpointcloudからeusを作る方は convex-hull-3d が点群によっては上手く変えてこないことがある気がします．https://github.com/jsk-ros-pkg/jsk_eus_pcl/blob/master/eus_pcl/euslisp/eus-pcl.l#L153　とか使えば少し良くなるのかな？？？これは味覚に．

YoheiKakiuchi, 2020-05-14:
メンテされてませんが、eus_pclにはkdtreeもあって、sample_poisson.lを一部修正するだけでそのまま使えます。
eus_pclのmelodicのコンパイルもできています。

octmapでモデル作ると、minecraftのモデルみたいのは簡単にできるかもしれないですね。

sample_poisson.lはuser::をeus_pcl::に修正
```
diff --git a/jsk/sample_poisson.l b/jsk/sample_poisson.l
index f3370617e..1058590e3 100644
--- a/jsk/sample_poisson.l
+++ b/jsk/sample_poisson.l
@@ -97,7 +97,7 @@
         (activefcmgr (if use-tree
                          (instance active-face-manager-tree :init obj2)
                        (instance active-face-manager :init obj2)))
-        (ktree (instance user::dynamic-kdtree :init))
+        (ktree (instance eus_pcl::dynamic-kdtree :init))
         points
         )
```

```
$ roseus
(load "package://eus_pcl/euslisp/eus-pcl.l")
(load "euslib/jsk/sample_poisson.l")

;; point-cloud
(setq pt (make-3dpointcloud-on-object-poisson (make-cube 100 100 100) 5.0 :irt t))

;; convex-hull (normalがおかしくて内向きのnormalが入っている/pclのバグなのか?)
(setq fs (eus_pcl::surface-construction pt))
```

YoheiKakiuchi, 2020-05-15:
たしかに以前octomapで色付きやってなかったので実装してみました。
https://github.com/jsk-ros-pkg/jsk_eus_pcl/pull/22

これがサンプルプログラムで、
https://github.com/jsk-ros-pkg/jsk_eus_pcl/blob/master/eus_pcl/test/create-octomap-from-eusmodel.l

以下、```(octomap-demo "kettle" 4.0)``` の結果です。

Eus model
[image]

Point cloud
[image]

octomap resolution 4mm
[image]

octomap resolution 8mm
[image]

octomap resolution 16mm
[image]

octomap resolution 32mm
[image]

octomap resolution 64mm
[image]