jsk-ros-pkg/jsk_roseus #655: サブスクライブしたメッセージから数字を取り出したい

k-okada, 2021-01-07:
```
もしくはサブスクライブしたメッセージのdataに含まれる文字列（"0"など）をeuslisp上で数値に変換する方法はあるでしょうか。

"0 57 371" というデータから
(parse-integer (subseq (send msg :data) 0 1)))
のように取り出すことをこころみましたが、parse-integerはcommon-lispでしか使えないでしょうか
```

kindsenior, 2021-01-07:
あまり状況が分かってないかもしれないので本末転倒の回答かもしれませんが，
(read-from-string "0")
とかでどうでしょう？

parse-intergerと違ってread-from-stringだと小数まで対応してしまうのでそれが嫌とかだと駄目とかはありそうですが．．．

k-okada, 2021-01-07:
以下のように`read-from-string` を使うと良いです．
```
1.eusgl$ (setq a "0 57 371")
"0 57 371"
2.eusgl$ (setq b (read-from-string (subseq a 0 1)))
0
3.eusgl$ (+ b 10)
10
```
また，S式を送ると以下のようにすることができます．
```
4.eusgl$ setq a "(list 0 57 371)"
"(list 0 57 371)"
5.eusgl$ (setq b (eval (read-from-string a)))
(0 57 371)
6.eusgl$ (+ (elt b 0) 10)
10
```

また，そもそも数字を送りたければ，そういうメッセージを送るのが良いです．例えば
```
$ rosmsg show geometry_msgs/Point
float64 x
float64 y
float64 z
```
を使って，`roscore` を起ち上げた後に
```
rostopic pub -r 1 /test geometry_msgs/Point   '{x: 10}'
```
としてパブリッシュして，
```
1.irteusgl$ (ros::roseus "test")
t
2.irteusgl$ (ros::subscribe "test" geometry_msgs::Point #'(lambda (msg) (print (+ (send msg :x) 10))))
t
3.irteusgl$ (ros::spin)
20
20
```
とすると数値としてデータを受け取ることができます．