jsk-ros-pkg/jsk_roseus #665: Ubuntu18ではPS3JOYの/joyトピックのボタン配置が違う

Naoki-Hiraoka, 2021-05-09:
```<node pkg="joy" type="joy_node" name="joy" />```でps3joyの`/joy`トピックを出していたのですが、

Ubuntu18では、ps3joyとのペアリング方法が変わったためか、
(http://wiki.ros.org/turtlebot_teleop/Tutorials/PS3%20Joystick%20Teleop のmelodicの欄参照)

`/joy`トピックのボタン配置が、これまでと変わりました。

https://github.com/jsk-ros-pkg/jsk_roseus/blob/b214284e24ad6287bf17545b9a73c541190fac1a/roseus/test/joy.l#L40
を変更したいのですが、他のubuntuのバージョンのロボットに影響を与えてしまいます。どうすべきでしょうか。

k-kimura, 2021-05-09:
@Naoki-Hiraoka とのubuntuアップデート作業で，単なる補足ですが
```
(load "package://roseus/test/joy.l")
(init-ps3joy)
```
などをしてるユーザ側のプログラムでは
`(send *joy* :square-button) ;; Ubuntu18環境ではこのボタン対応付けが変わってる`
を
`(elt (send *joy* :buttons) 3) ;; このeltの番号をボタンに応じて変える`
みたいな修正になってます．

708yamaguchi, 2021-05-09:
roseusレベルでの解決ではないので参考程度ですが、2台のfetchでそれぞれindigoとmelodicを動かしていたときは`joy_remap.py`というノードを使って`/joy`トピックの配列の順番を変更していました。

https://github.com/jsk-ros-pkg/jsk_robot/pull/1317
https://github.com/ros-drivers/joystick_drivers/blob/master/joy/scripts/joy_remap.py

ketaro-m, 2022-11-15:
@708yamaguchi こちらで追加されている2つのyamlファイルですが，[本家](https://github.com/ros-drivers/joystick_drivers/tree/master/joy/config)にPRを出した方が良いと思うですがどうでしょうか？
indigoは試していないですが，melodicのyamlファイルは正しく動作することを僕の方でも確認しました．

708yamaguchi, 2022-11-15:
@ketaro-m 
本家にPRを出すのはすごく良いことだと思うので、お願いしてもいいでしょうか。（以前僕がやっておくべきでした、すみません。）

indigoで動作確認できていないのは仕方ないと思うので、気にしなくても良いと思います。
今のros-drivers/joystick_driversのmasterブランチでもindigoは考慮されていないと思います。