jsk-ros-pkg/jsk_roseus #671: What is the right way to define preemptable action servers?

Affonso-Gui, 2021-06-15:
To accept  `/cancel` messages from other topics during the execution of a `ros::simple-action-server` callback I had to use something like the following.

```lisp
(defun execute-cb (server goal)
  ;; ...

  ;; Check for interruption
  (send server :spin-once)
  (when (send server :is-preempt-requested)
    (send server :set-preempted)
    (return-from execute-cb nil))

  ;; ...
)
```

Am I missing something here or is this actually the right way to do it?

Also I could not find any solid examples for the action server implementation, do we have any?

k-okada, 2021-06-15:
> Also I could not find any solid examples for the action server implementation, do we have any?

that's true, is there any good example of action server written in other languages? we will start with translating that code to euslisp
c.f. https://github.com/jsk-ros-pkg/jsk_roseus/blame/b214284e24ad6287bf17545b9a73c541190fac1a/roseus/test/add-two-ints-server.l#L3

Affonso-Gui, 2021-06-15:
Maybe this one? http://wiki.ros.org/actionlib_tutorials/Tutorials/SimpleActionServer%28ExecuteCallbackMethod%29

It is similar to the one implemented in https://github.com/jsk-ros-pkg/jsk_roseus/blob/b214284e24ad6287bf17545b9a73c541190fac1a/roseus/test/fibonacci-server.l , but it can be interrupted.

Should I upgrade the example and send a PR?

k-okada, 2021-06-16:
> Should I upgrade the example and send a PR?

yes please

Affonso-Gui, 2021-06-17:
PR created, but I still think it would be nice if the sample had a bit more visibility, maybe with some documentation similar to http://wiki.ros.org/ja/ROS/Tutorials/WritingServiceClient%28euslisp%29 or included in the roseus_tutorials?