jsk-ros-pkg/jsk_roseus #674: rad表記のfloat-vectorをdegに変換する方法について

softyanija, 2021-06-18:
タイトル通りradで書かれたfloat-vectorをまるごとdegに変換したいのですが、一発でそのような変換を行ってくれる関数が調べても見つかりません。山口さん(@708yamaguchi)に質問したところ以下のようにするといいと教えていただきました。
```
(setq joint-states (coerce (mapcar #'(lambda (x) (rad2deg x)) (coerce *\
joints* cons)) float-vector))   
```
ここではrad表記の*joints*をdeg表記の joint-states に変換しています。
他に良い方法が合ったら教えていただけると幸いです。

Naoki-Hiraoka, 2021-06-18:
上の例と大差ありませんが，
`(map float-vector #'rad2deg #F(1 2 3))`
でできます．

k-okada, 2021-06-18:
ちなみに，これ，どこで必要なのかな．

eusの世界でプログラムを書く時点では degで扱いたいので，
ROSでradで出てきた情報は，*ri* から取り出す時点で，deg になっているのが望ましいです．
https://github.com/jsk-ros-pkg/jsk_robot/blob/25ac99ab82539b0e95d20470db1675b5f2f58c56/jsk_naoqi_robot/naoqieus/naoqi-interface.l#L125

softyanija, 2021-06-18:
@Naoki-Hiraoka 
短くて楽ですね。ありがとうございます。

@k-okada 
83C1のdual_pandaだと/dual_panda/joint_statesという形でjointの情報が出ていて、そこから :position を取ってくるとrad表記になっていました。
実機に繋いで (send *rI* :potentio-vector) のようにするとdegで出てきますが、rosbagに対しての使用のためにこのようにしています。

k-okada, 2021-06-18:
ROSのクライアントからみたら，データを取得するうえでは実機でもrosbagでも同じなのでrosbag からでも *ri*
:potentio-vector できるはずです．わからなかったらrosbag 送って下さい．
--
◉ Kei Okada

softyanija, 2021-06-18:
https://github.com/softyanija/jsk_robot/blob/add_panda_robot/jsk_panda_robot/panda_eus/euslisp/end-coords_publisher.l
このプログラムでの話で、(dual_panda-init) で *ri* を作ろうとすると下のメッセージのようにactoin serverが見つからないと言われます
```
74.irteusgl$ (dual_panda-init)                                              
[ WARN] [1624007550.969777206, 1623906081.179971108]: topic /robot_interfac\
e_marker_array already advertised                                           
[ WARN] [1624007550.973012407, 1623906081.179971108]: topic /dual_panda/dua\
l_panda_effort_joint_trajectory_controller/follow_joint_trajectory/goal alr\
eady advertised                                                             
[ WARN] [1624007550.973038241, 1623906081.179971108]: topic /dual_panda/dua\
l_panda_effort_joint_trajectory_controller/follow_joint_trajectory/cancel a\
lready advertised                                                           
[ WARN] [1624007550.976105153, 1623906081.189986647]: Waiting for actionlib\
 interface forever because controller-timeout is nil                        
[ WARN] [1624007556.078800648, 1623906086.295045299]: Still waiting for [/d\
ual_panda/dual_panda_effort_joint_trajectory_controller/follow_joint_trajec\
tory] action server. 5 seconds have passed.
```
rosbagはメールで送ります。このブランチの状態で動かしてます。https://github.com/softyanija/jsk_robot/tree/add_panda_robot

softyanija, 2021-06-18:
[ここ](https://drive.google.com/file/d/1jlGVaAFMPu-qu8_sADgBGZIGqDpN2J2Q/view?usp=sharing)にrosbagを置いておきました

k-okada, 2021-06-18:
https://github.com/ykawamura96/jsk_robot/pull/4/commits/b6c87f2981d20ae9a6dbf7e1a1cc9bf7d4cd8f04 をして
```
(load "package://panda_eus/euslisp/dual_panda-interface.l")

(defmethod dual_panda-robot-interface
  (:dummy-controller (&rest args)))
(setq *ri* (instance dual_panda-robot-interface :init
                     :type :dummy-controller))

(do-until-key
 (print (send *ri* :state :potentio-vector))
```
でbagfile の情報が　`(send *ri* :state ) `に反映されるとおもいます．

ただ，コレ，ソモソモは，手先の位置をグラフにしたいんだよね？

[image]
で良い気がするけどどうだろう．見ても，グラフに変化がないけど，ちなみに送ってくれたBagファイルはロボットは動いていない，とういのでいいのかな．

```
sudo apt install ros-melodic-plotjuggler-ros
```
したあと，

```
rosrun plotjuggler plotjuggler
```
でいけるはず．

softyanija, 2021-06-23:
画像などの情報まで使うようになると時間的に微妙に正確でないことが効いてきそうなのでそれまでに改良したいですね。他のタスクが多く手が回りきらないので後ほど試して報告します。

YoheiKakiuchi, 2021-06-28:
問題の本質とはずれてる気もしますが、
以下だと実行速度、メモリ使用(GC回避)ともに良いと思います。
(joint-statesはfloat-vectorと仮定していて、内容が変更されます)

```
(scale (rad2deg 1.0) joint-states joint-states)
```

@softyanija @Naoki-Hiraoka 
の結果は正しいですし、意味的な読みやすさはそちらの方が良いかと思います。
僕のはeus初期のCPUが貧弱だったころのやり方ですね。