jsk-ros-pkg/jsk_roseus #704: ServiceClientオブジェクトのmethodが上手く定義されない

Kanazawanaoaki, 2022-02-03:
```
$ rossrv show pr2_mechanism_msgs/SwitchController 
int32 BEST_EFFORT=1
int32 STRICT=2
string[] start_controllers
string[] stop_controllers
int32 strictness
---
bool ok

```
のサービスのServiceClientオブジェクトを生成してもstart_controllersのmethodが定義されない．
```
$ roseus
1.irteusgl$ (ros::load-ros-manifest "pr2_mechanism_msgs")
t
2.irteusgl$ (setq req (instance pr2_mechanism_msgs::SwitchController :init ))
#<pr2_mechanism_msgs::switchcontroller #X563e2d19dc90>
3.irteusgl$ (send req :start_controllers "r_arm_controller_loose")
Call Stack (max depth: 20):
  0: at (send req :start_controllers "r_arm_controller_loose")
  1: at #<compiled-code #X563e2c217e18>
/opt/ros/melodic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: cannot find method :start_controllers in (send req :start_controllers "r_arm_controller_loose")
4.E1-irteusgl$ (send req :methods)
(:init :md5sum- :datatype- :connection-header :plist :get :put :name :remprop :prin1 :prin1 :warning :error :slots :methods :super :get-val :set-val)
```

例えば他のサービスを試してみると，
```
1.irteusgl$ (ros::load-ros-manifest "dynamic_tf_publisher")
t
2.irteusgl$  (setq req (instance dynamic_tf_publisher::SetDynamicTFRequest :init))
#<dynamic_tf_publisher::setdynamictfrequest #X562123876c30>
3.irteusgl$ (send req :methods)
(:init :freq :cur_tf :serialization-length :serialize :deserialize :response :init :md5sum- :datatype- :connection-header :plist :get :put :name :remprop :prin1 :prin1 :warning :error :slots :methods :super :get-val :set-val)
```
のようにmethodが定義されています．

pythonだと，
```
#!/usr/bin/env python
# -*- coding: utf-8 -*-

import rospy
from pr2_mechanism_msgs.srv import SwitchController

sp = rospy.ServiceProxy('/pr2_controller_manager/switch_controller', SwitchController)
resp = sp(start_controllers=["r_arm_controller_loose"], stop_controllers=["r_arm_controller"])
```
のように書くことで実行できることは確認しました．  

何が問題なのでしょうか？

Kanazawanaoaki, 2022-02-03:
```
(setq req (instance pr2_mechanism_msgs::SwitchController :init ))
```
を
```
(setq req (instance pr2_mechanism_msgs::SwitchControllerRequest :init ))
```
と`Request`を追加したものに変更したら上手く定義されました．
```
1.irteusgl$ (ros::load-ros-manifest "pr2_mechanism_msgs")
t
2.irteusgl$ (setq req (instance pr2_mechanism_msgs::SwitchControllerRequest :init ))
#<pr2_mechanism_msgs::switchcontrollerrequest #X55a56f9bc410>
3.irteusgl$ (send req :methods)
(:init :start_controllers :stop_controllers :strictness :serialization-length :serialize :deserialize :response :init :md5sum- :datatype- :connection-header :plist :get :put :name :remprop :prin1 :prin1 :warning :error :slots :methods :super :get-val :set-val)
```

Affonso-Gui, 2022-02-03:
Looks a lot like https://github.com/jsk-ros-pkg/jsk_roseus/issues/543 https://github.com/jsk-ros-pkg/jsk_roseus/pull/546

Should we add a wrapper to this one too?