jsk-ros-pkg/jsk_roseus #709: Euslispでyaml形式のファイルを読み書きしたい

Kanazawanaoaki, 2022-04-30:
Euslispでyaml形式のファイルを読み書きするための関数や読み書きしている例はありますでしょうか？
  
[python](https://github.com/yaml/pyyaml )や[common lisp](https://github.com/eudoxia0/cl-yaml )の例はあるみたいなのでそれらを参考にして作るか，使いたいyamlファイルの形式などがある程度決まっている場合は自分でparserを作るというのも有り得そうです．  

また，読み込むだけならrosparamとして渡すという作戦が有効そうかなと思っています．

因みにjsonの例は[jsk_roseus/roseus_mongo/euslisp/json/]( https://github.com/jsk-ros-pkg/jsk_roseus/tree/master/roseus_mongo/euslisp/json)にあるようです．

Affonso-Gui, 2022-05-01:
As far as I know the closest thing we have to a yaml utility in euslisp is the rosparam interface.

You could try to couple set/get param with `(unix:system "rosparam load/dump ...")` calls, or write a simple encoder /decoder and maybe add it to the euslib or roseus-utils.

Here is a start:
```lisp
(defun alistp (alist)
  (and (listp alist) (every #'consp alist)))

(defun format-alist (lst output-stream &optional (lvl 0))
  (dolist (l lst)
    (let ((name (car l))
          (val (cdr l)))
      (spaces (* 2 lvl) output-stream)
      (format output-stream "~a: " name)
      (if (alistp val)
          (progn
            (terpri output-stream)
            (format-alist val output-stream (1+ lvl)))
        (if (listp val)
            (dolist (v val)
              (terpri output-stream)
              (spaces (* 2 (1+ lvl)) output-stream)
              (format output-stream "- ~A" v))
          (print val output-stream))))))

;; (format-alist (ros::get-param "myparam") t)
```

Kanazawanaoaki, 2022-05-12:
Thank you very much.  
I confirmed that I can load and dump yaml from eus that I wanted to do.  

I feel that simple encoder/decoder can be done relatively easily if we use rosparam, so I'll think about it for a moment.

input.yaml
```
val1: 100
val2: 200
```
from-yaml.l
```
#!/usr/bin/env roseus

(ros::roseus "test")

(defun yaml-load (input-file)
  (unix:system "rosparam delete /yaml_test")
  (unix:system (format nil "rosparam load ~A yaml_test" input-file))
  (setq val1 (ros::get-param "/yaml_test/val1"))
  (setq val2 (ros::get-param "/yaml_test/val2"))
  (format t "~%load params from ~A ~%" input-file)
  (format t " now val1 is ~A ~%" val1)
  (format t " now val2 is ~A ~%" val2)
  )

(defun yaml-dump (output-file)
  (unix:system (format nil "rosparam dump ~A yaml_test" output-file))
  (format t "~%dump params to ~A ~%" output-file)
  )

(yaml-load "input.yaml")
(yaml-dump "output.yaml")
```
in command line
```
$ ls
from-yaml.l  input.yaml
$ roseus from-yaml.l 

load params from input.yaml 
 now val1 is 100.0 
 now val2 is 200.0 

dump params to output.yaml 
$ cat output.yaml 
val1: 100
val2: 200
```

Kanazawanaoaki, 2022-05-19:
Using https://github.com/jsk-ros-pkg/jsk_roseus/pull/710 , I created a program to read and write any yaml file in roseus by using the load and dump functions of rosparam.
https://gist.github.com/Kanazawanaoaki/b9aba2c4bb1a51e138057574beab7dfe  
```
(defun yaml-load (input-file)
  (unix:system "rosparam delete /eus_yaml_load")
  (unix:system (format nil "rosparam load ~A eus_yaml_load" input-file))
  (setq *params* (ros::get-param "/eus_yaml_load"))
  (unix:system "rosparam delete /eus_yaml_load")
  (format t "~%load params from ~A~%" input-file)
  (format t " now param is ~A ~%" *params*)
  )

(defun yaml-dump (output-file)
  (unix:system "rosparam delete /eus_yaml_dump")
  (ros::set-param "/eus_yaml_dump" *params*)
  (unix:system (format nil "rosparam dump ~A eus_yaml_dump" output-file))
  (unix:system "rosparam delete /eus_yaml_dump")
  (format t "~%dump params to ~A~%" output-file)
  )
```
Is there a good way and need to add such functions to roseus?

Affonso-Gui, 2022-05-30:
Glad that you were able to read/ write yaml files as expected.

However, I am afraid that this still may be a little too 'hacky' to add to jsk_roseus...

Kanazawanaoaki, 2022-05-30:
That's right ...
I don't think it's something that can be added to jsk_roseus either.
Is it better to leave it as a sample program in this issue?
I have no idea how to add it to jsk_roseus as a feature.

Affonso-Gui, 2022-05-30:
I think it is ok to leave it as a sample program in this issue.